teachingtechYT/teachingtechYT.github.io #1: [Feature Request] Needs More Nozzle Size support

Zogar89, 2020-08-14:
Many of us use nozzle sizes larger than the standard 0.4mm bore.
Even in those unknown sizes where there are no good parameter guides, experimentation and calibration becomes more necessary.

Andychud, 2020-08-15:
Agree 100%

teachingtechYT, 2020-08-15:
This was considered and will probably be added. I'll have selector that states nozzle size and layer height eg. 0.4mm nozzle @ 0.2mm layer height.
I wouldn't plan to have millions of combinations however.
Probably just a common layer height for each nozzle.
What do you guys consider to be standard for 0.2, 0.6, 0.98 and 1.0mm?

jagdeepsingh02, 2020-08-15:
> o you guys consider to be standard for 0.2, 0.

Hi, i think it's better to add different layer height especially for retraction a temperature tower test, in my case same roll of filament needs different retraction and different nozzle temperature to print correctly..
thanks for the huge effort for the community by the way.

Andychud, 2020-08-16:
> This was considered and will probably be added. I'll have selector that states nozzle size and layer height eg. 0.4mm nozzle @ 0.2mm layer height.
> I wouldn't plan to have millions of combinations [...]

Yes, In order to keep things simple and manageable I'd just look at including "common" nozzle/layer height settings. This would cater for the vast majority of users. If anybody is using settings close but not exactly the same then the results should be "close enough" for them. I'd give the following options...

Nozzle    Layer Height
0.2          0.1
0.4          0.2
0.6          0.3
0.8          0.4
1.0          0.5

EccentricVamp, 2020-08-18:
Just wanted to bump this request and voice my support. Also, the Nozzle/Height options that @Andychud mentioned look good.
P.S. Github supports Markdown tables:
| Nozzle Diameter | Layer Height |
|-----------------|--------------|
| 0.2 mm          | 0.1 mm       |
| 0.4 mm          | 0.2 mm       |
| 0.6 mm          | 0.3 mm       |
| 0.8 mm          | 0.4 mm       |
| 1.0 mm          | 0.5 mm       |

AndrezinhoCosta, 2020-08-25:
I was looking for this feature! Many nozzle diameters and layer heights.

kleverline, 2020-08-25:
I second what EccentricVamp put.   5 options, each with a set layer height. I use 0.6mm and these tools would be extremely helpful if I could adjust them to match!

pressionllc, 2020-09-11:
Just another vote for this option. I run the CR10s Pro for big prints and I don't use anything smaller than 0.6 nozzle. I think the table of nozzle and layer heights proposed by @EccentricVamp make good sense.

Great tool! I'm eager to be able to apply it!

damien6669, 2020-10-23:
This would be amazing - I am currently testing different nozzle sizes, and this would be a great resource for setting them up!

teachingtechYT, 2021-07-13:
This very highly requested feature is being added this week. Thanks for the discussion.

closed 2021-07-13 by teachingtechYT