teachingtechYT/teachingtechYT.github.io #2: [Feature Request] Disable Z Hop on retraction tests

Zogar89, 2020-08-14:
Z Hop motion can be unnerving. Usually generates many threads and give the impression that a value higher retraction is needed.
Disable Z Hop is a must.

Andychud, 2020-08-14:
> Z Hop motion can be unnerving. Usually generates many threads and give the impression that a value higher retraction is needed.
> Disable Z Hop is a must.

Agree with this!

teachingtechYT, 2020-08-15:
I originally wrote the retraction generator to allow you to alter the existing three parameters plus z hop, but it proved to be quite complicated to implement.
I could manually remove the z hop lines from the source gcode, or potentially flag them so they could be commented out.
This would give the option for no z hop or whatever it is now 0.2mm I think).

dmarx, 2020-08-18:
_(Using an ender 3 pro)_

As someone new to 3D printing (only about a month in), I've been relying on cura profiles other people have developed. None of the default cura profiles have z-hop enabled, and neither do any of the ones I've imported from other content creators (CHEP and FDG). I stumbled on these profiles because they seem to be recommended a lot, so I imagine a lot of ender 3 users don't normally print with z-hop enabled.

teachingtechYT, 2020-08-19:
Currently disabled on the retraction test (with the other to come). Soon users will be able to set Z hop along with existing parameters.

closed 2020-08-19 by teachingtechYT