teachingtechYT/teachingtechYT.github.io #4: [New Calibration] X,Y,Z Step

OlivierMary, 2020-08-14:
Hi there,

Very good calibrations pages. And first thanks a lot for this !!

Maybe an useful other calibration test is for X,Y and Z steps motors. => M92 + M500
With xyz cube or other xyz calibration models more efficient with bigger length.

teachingtechYT, 2020-08-15:
I considered this, and might add it in time.
I was concerned that people would think it was an essential step whereas in reality most printers won't need any changes. A don't fix it if it ain't broke scenario.

BaelfireNightshd, 2020-08-15:
Also, from what I've read, unless the x, y, or z steps are wrong out of box (or you've modified something), the problem causing different dimensions on parts is usually over/under extrusion. You can test this by checking the error on a 20mm cube vs a 30mm cube. Then turn it into a linear relationship (y=mx+b, with x being ideal, and y being actual dimension) and you'll find the m is super close to 1, but the b is where most of your error comes from (should be 0 on a perfect printer). If you need clarification, met me know.

OlivierMary, 2020-08-15:
If X,Y and Z motors are always good from start why E not ? If change motherboard or flash marlin you have to know how to set the goods motors steps, maybe warn people to do that in last calibration check if all other check are good and X,Y,Z cube not good.

vector76, 2020-08-15:
A small-radius hobbed gear in an extruder can much more easily have a few percent error, while such errors in X, Y, and Z are less likely.  Belts can stretch slightly but this is very small relatively speaking.  Even the idler tension in an extruder could slightly change the (effective) gear radius, so it's hard to nail the value exactly from the initial design.

Not only are XYZ usually pretty accurate to start with, naive XYZ measurements can easily make the problem worse.  The procedure suggested by @BaelfireNightshd is a good one (though I might suggest a 20 mm cube vs. a 120 mm cube).

OlivierMary, 2020-08-15:
Ok will check with this test will see :)

BaelfireNightshd, 2020-08-15:
To add onto what @vector76 mentioned, your filament can actually change the E steps too (though you usually just compensate with flow rates) since the hobbed gear will dig into different filaments different amounts. In short, there's a lot more variables that go into the E steps than the x, y, or z steps.

teachingtechYT, 2020-08-24:
My thoughts, which are already at the top of the E-steps page:

> For the X, Y, and Z axes, the steps per mm is usually consistent between printers and rarely changes with modifications. As long as belts are tight and true, it rarely needs to be tuned.
> For the extruder however, variations in extruder hardware and filament means it is worth properly calibrating the extruder steps per mm, or E-steps.

Still willing to add this test down the track, but there is going to be a lot of special notes to cover the range of factors that can affect the dimensions being very slightly off.

TheAndr0id, 2020-08-30:
I found this in the comments for the Thingiverse entry for the XYZ Cube (https://www.thingiverse.com/thing:1278865).  It's simple and slick.

> 
> Doril81
> May 31, 2020
> 
> Thank you for the cube! it was really helpful to calibrate my printer!
> 
> I created a small calculator for your cube at https://3dpc.nl/

vector76, 2020-08-30:
Small cubes are unreliable for size calibration because over- or under-extrusion can influence the dimensions by a significant amount, leading to incorrect 'calibration'.  It is common for the 'calibrated' values to be worse than uncalibrated values.

A better test would use a large, hollow square or similar, or would use a differential measurement that is mostly invariant to extrusion errors.

swordfish45, 2020-09-05:
@TheAndr0id 

That is a neat tool, but has to be used carefully. 

You shouldn't be relying on prints to do x,y,z calibration at all. There are other variables that affect print dimensions, and for kit printers, you should never need to touch x,y,z steps. 

For example, if the test print came out fine with x and y dimensions, but short in z, one would think to up the z steps accordingly. But if you lost z due to nozzle too close to bed, and excessive elephants foot, you have adjusted a multiplier to overcome a constant error. 

If you must, just use calipers, don't rely on print dimensions.

teachingtechYT, 2020-09-06:
A long time ago I used to mount digital calipers against the print head, send a G1 move and then look at the before and after reading on the calipers.
The same thing can be done with a dial gauge or with less accuracy with a ruler.

TheAndr0id, 2020-09-07:
THANK YOU!  These kind of ideas/experiences are why I'm here...  Searching Thingiverse now (at least until "Something Went Wrong" happens....)

Yes, I can see how manually pushing a measuring device could lead to measurement accuracy, but I doubt it will be measurable with my calipers.  This train of thought opens many thoughts around xyz calibration as well as extruder calibration as well!

This is the kind of calibration I like to see.  Something quick, doesn't waste filament with useless cube prints and can be done quickly after any adjustments are made to the printing mechanics!

TheAndr0id, 2020-09-11:
I have results!

So first off I needed some tools to do this.  I used the "cross" from the first thing and the mounting bracket from the second.  The cross worked well after bending it in a little to make a snug fit.  The mounting bracket?  It worked, but I wouldn't trust it to hold the calipers that long...

https://www.thingiverse.com/thing:2237494   (I just used the cross piece, but the "holder" also looks usefull)
https://www.thingiverse.com/thing:3044827

Now using these two printed items attached to my calipers and bolted to my Ender 3V2, I was able to get some good solid measurements.  You could easily do all this from a console, but I decided to write some GCode (I said I wasn't going to do this) so I could just "print" the calibration steps.

I did 6 runs of the X, Y and Z and got solid measurements within 0.01-0.04 of each other.  Here's my results for the sake of data: (X and Z 100mm tests, Y is only 75mm since there wasn't space for the caliper and 100mm)

X: 99.67, 99.69, 99.67, 99.68, 99.69, 99,69  AVG: 99.68
Y: 74.82, 74.82, 74.85, 74.86, 74.86, 74.83  AVG: 74.84
Z: 99.97, 99.98, 99.99, 99.96, 100.00, 100.01  AVG: 99.985

Now this doesn't require printing anything and took around 20 minutes to do all 18 tests.  I spent more time tweaking my GCode to get the beeps working the way I wanted them... :)

I've included my X calibration GCode. (Change X to Y or Z and don't forget to change the 100mm to 75mm for Y!)

```
;
; Let's do all of our movements in mm.  The movements should be relative 
; movements as well.
;
G21
G91

;
; In order to make sure we're not hitting up against the limits, move the X, Y 
; and Z axis to the home position then lift the print head 10mm just to make 
; sure we're not hitting the bed.  
;
G28
M400
G0 Z10 F1500
M400

;
; Play a beep and then move the X 5mm to make sure the caliper is resting 
; against the print head/frame/bed.  
;
M300 P200
G4 S5
G0 X5 F750
M400

;
; Play 2 beeps and move the X axis 100mm.
;
M300 P200
G4 P200
M300 P200
G4 S5
M117 Moving X 100mm
G0 X100 F750
M400

;
; Play a tone to say we're done and move the head back 10mm.  After waiting
; for it to arrive, wait another second and auto home.  Shut off the fans and 
; wait 5 seconds for the LCD display to catch up.
;
M300 P200
G4 S1
G0 X-10 F750
M400
G4 S1
M117 Completed
G28
M400
M107
G4 S5
```
How to use:

1. Mount caliper to frame or hold it *REAL* tight.
2. "Print" this file.
3. At first beep, move caliper arm up to the print head/frame/bed.
4. At the double beep reset your caliper to zero.
5. The final beep take your measurement.
6. Record the result and repeat as needed.

There was only one "gotcha".  I had to slow down the movement rate as my caliper kept losing it's marbles and jumping to weird values.  It's cheap, but it *mostly* works.  Mostly...

BaelfireNightshd, 2020-09-11:
That's really cool that you compiled that. One thing I'd add to the test is actually moving different amounts to account for any absolute error vs relative error. If you're adjusting steps that's a relative setting so you want to eliminate any possible absolute error influencing your calibration. For example, I know lead screws typically have backlash that is an absolute error. If it's an error of .5 of a mm (making up numbers) when you move 20mm, it'll still be .5 when you move 40mm.

TheAndr0id, 2020-09-12:
Ok so your saying doing tests at different lengths?  Aka:  X at 100mm, then at 80mm then at 60mm?

I really don't know what I'd do with such data.  I guess you could check to see if it was consistent and if not it might give you an indication there was an extra error.  How would you determine what that was?

I'll be honest, I'm getting tired of calibrating my printer.  I just want to print some cool shit.  :)

BaelfireNightshd, 2020-09-12:
That's perfectly reasonable and what I described above is probably overkill.

However, to answer what you'd do with that data, I would use two data points and make a linear equation (like y= m*x + b) to split it into relative error (m should be 1 if perfect) and absolute error (b should be 0 if perfect). 

So for example, let's say I took 3 measurements at 100mm and 3 measurements at 60mm and then took the average of each set. Let's call the two averages Y1 and Y2 respectively. X1 and X2 would be 100mm and 60mm respectively. 

Now to find m we need to plug the numbers into the equation m = (Y2 - Y1) / (X2 - X1). The difference between m and 1 is the relative error. 

Now what we can do is take my steps lets say 100 and multiply it by 1/m for my new steps. 

A possible better way to update steps is inspired by gradient descent and learning rates in Machine Learning (don't run away. This is an easy concept). One problem you can run into while doing the above is you can bypass the right step amount so far that you end up with bigger and bigger oscillations out of calibration. So instead of 1/m, I use 1 / [(m - 1) * 0.5 + 1]. This takes half steps in the direction pointed to by m. This of course will require several calibrations, but it should converge on the correct steps eventually / you'll say that's good enough. 

Any difference between b and 0 is from some other absolute error cause such as over/under extrusion or backlash. 

Ask any questions if you have them or let me know if you can think of any improvements to my math/process. Again. This is probably overkill.

plank3ffect, 2020-12-30:
Actually I support adding these steps if possible.  With my Biqu B1, my X axis step was off.  It was 20.3mm instead of 20mm.  Y and Z were spot on.  Adding this would be great.  

Also, it looks like the X and Y axis are reversed on the generated cube gcode?  When adjusting M92 X, the Y axis on the cube was affected.  Perhaps rotate the cube 90 degrees to swap X and Y axes?

teachingtechYT, 2021-07-13:
This one was added a while ago.

closed 2021-07-13 by teachingtechYT