teachingtechYT/teachingtechYT.github.io #8: Modifiers for different nozzle sizes.

jhenley01, 2020-08-14:
Hello,

This is an awesome project, thank you for creating this. I mostly use a 0.6 nozzle so some of the tests did not print correctly for me, it would be awesome to have options for different nozzle sizes. Thanks again!

Andychud, 2020-08-14:
I use 0.6 too and agree 100% that there should be option to select different nozzle sizes.

Andychud, 2020-08-15:
I just noticed that this has already been requested/reported
https://github.com/teachingtechYT/teachingtechYT.github.io/issues/1#issue-679402188

jhenley01, 2020-08-15:
> I just noticed that this has already been requested/reported
> [#1 (comment)](https://github.com/teachingtechYT/teachingtechYT.github.io/issues/1#issue-679402188)

Ah good call, somehow I did not see that. Closing.

closed 2020-08-15 by jhenley01