teachingtechYT/teachingtechYT.github.io #10: Combine temperature and rollback calibration

zakharsk, 2020-08-15:
These [guys](https://www.cnx-software.com/2020/07/08/3d-printer-retraction-calibration-vol-ii-calibration-generator-program-release/) did a great job and created the GCode generator that allows you to adjust the temperature and retraction at the same time. [on-line version](http://prahjister.duckdns.org/)

The problem with their code is that it must be fully executed two times: 

1. variable temperature and retraction distance
2. variable retraction speed, but fixed temperature and retraction distance

If for the second time there was a reduced tower (20x20 for example), it would cut a lot of time.

teachingtechYT, 2020-08-15:
It looks very powerful, probably more so than this.
I wonder about how daunting it would be for beginners. Already I have 15 fields for the retraction part and I was worried it could be confusing.
The correct balance is important.

rbarreiros, 2020-08-15:
Could I suggest a switch between easy version and expert, where in the easy you just select the begin/end values and steps, example, start at 190º end at 210, 5º steps and the code does the rest, then an expert version with further customization.

teachingtechYT, 2020-08-17:
I've added the link to the online version to the bottom of the retraction page so people have a choice.

closed 2020-08-17 by teachingtechYT