teachingtechYT/teachingtechYT.github.io #12: Add an option for a purge line on the left side of the bed

smellott, 2020-08-15:
I always have Cura do a purge line prior to starting a print to get the filament flowing and use the skirt to see if I need to adjust baby steps.

teachingtechYT, 2020-08-15:
I purposed removed this as it won't work for delta printers. I left in the skirt for all prints except the acceleration test as this put it outside my 100 x 100mm baed target.

brainbukt, 2020-08-15:
how about doing a head lift to a set z height and perfoming a mid air purge that can be cleared by the user?

teachingtechYT, 2020-08-17:
I feel it adds unneeded complexity for very little gain in performance. If the user is present they can just as easily remove any dodgy bits from the build surface while the skirt is printing.

MattFryer, 2020-08-17:
How about just adding more skirts? I've been printing the retraction tuning and I've had the first layer print with no perimeters because the nozzle wasn't primed.

teachingtechYT, 2020-08-18:
Custom start gcode can now be added, which means you can add in purge in any manner you like. Soon this should be persistent so very user friendly.

closed 2020-08-18 by teachingtechYT

MattFryer, 2020-08-18:
Good solution!