teachingtechYT/teachingtechYT.github.io #13: Adding measurements setting

siryone, 2020-08-15:
Hi. Can you add ways to get the printed results accurate?
Such as control over expansion and hole sizes

teachingtechYT, 2020-08-15:
It would be very difficult. I have to start with normal gcode and then modify specific parts using javascript. Things like the actual flow rate are hard coded in and it would be very complicated to modify on the fly.
Not exactly what you want but most printers allow you to adjust speed and flow rate mid print from the LCD.

siryone, 2020-08-15:
Changing the flow rate while printing will not help since I can not measure during this stage

teachingtechYT, 2020-08-16:
The issue is that some slicers offer compensation and others do not. It becomes murky and confusing for beginners if I'm suggesting something that doesn't apply to them.

closed 2020-08-20 by teachingtechYT