teachingtechYT/teachingtechYT.github.io #15: Add leveling/distamnce print

rendez2k, 2020-08-15:
Hi there, fantastic idea. One thing I always do is print 5 squares/circles on my bed to check leveling as per this: https://www.youtube.com/watch?v=RZRY6kunAvs&list=PLyelYMwIkPd4SlwcBTDj92t0eyGpvzIkB&index=2&t=196s

I find it an important step to double check leveling and distance

teachingtechYT, 2020-08-16:
I agree this would be a good thing to include. I usually use an 'X' but this would be easier to generate gcode for. It's still complicated as the entire gcode won't be moved together as the current generators do, but rather the five squares need to be positioned independent of each other in the right proportion. The delta version will be a mess too.
This will need time to get right.

teachingtechYT, 2020-08-24:
This is now implemented, suits rectangular bed and delta bed with specified diameter. Full page with video snippets and refence images and diagrams.
Thanks to vector76 for their help.

closed 2020-08-24 by teachingtechYT