teachingtechYT/teachingtechYT.github.io #16: Include G90 - Command after using G91

rupertk79, 2020-08-15:
I followed the steps of the Extruder calibration and I've forgotten to set to absolute mode again. 

Thanks

brainbukt, 2020-08-15:
alot of start gcode actually sets extruder mode to absolute anyway

fredrikbaberg, 2020-08-15:
Would this be why my "slicer flow calibration" cube tried to print outside the "top" right corner, printed after calibrating e-steps according to instruction?

rupertk79, 2020-08-15:
> Would this be why my "slicer flow calibration" cube tried to print outside the "top" right corner, printed after calibrating e-steps according to instruction?

I think so. That‘s what I‘ve been watching

fredrikbaberg, 2020-08-15:
> > Would this be why my "slicer flow calibration" cube tried to print outside the "top" right corner, printed after calibrating e-steps according to instruction?
> 
> I think so. That‘s what I‘ve been watching
> After a G90 it was fine

I did a power cycle which also helped, but if a G90 is enough that would be better.
Just created a pull request, will see if it is accepted as is or if there are other ideas.

teachingtechYT, 2020-08-16:
I agree it is safer to have G90, I have ammended.

closed 2020-08-16 by teachingtechYT