teachingtechYT/teachingtechYT.github.io #17: Extruder steps/mm must be calibrated without load

PKav89, 2020-08-15:
Extruder E-steps Calibration page says "Now heat up your hot end to whatever temperature you usually print with."
This is incorrect. If extruder is under load, fillament will slip over the gear slightly and you will get wrong results, e.g. different results for different fillaments or different gear tension, if extruder is capable of adjusting it (like BMG).
In case of bowden setup calibration can be done with bowden tube detached.
In case of direct setup, the calibration can be done with backwards movement or with dismounted extruder.

astupidmoose, 2020-08-15:
> Extruder E-steps Calibration page says "Now heat up your hot end to whatever temperature you usually print with."
> This is incorrect. If extruder is under load, fillament will slip over the gear [...]

Even with a Bowden detached, printer firmware often requires a heated nozzle in order to preform this calibration as it won't allow you to push the extruder if the nozzle is cold...

vector76, 2020-08-15:
Do you not want to measure the actual travel under a realistic load?  I'm trying to understand the advantage to measuring an idealized travel when not under load.  I agree it could be different.

PKav89, 2020-08-15:
@vector76 yes, the idealized fillament travel is required here . It will also be used during retraction movements, during fillament change routine, linear advance algorithm, etc. There is another, fillament-specific parameter, that is used under load - flow multiplier, which is calibrated individually for every fillament after E-steps was set correctly.

teachingtechYT, 2020-08-16:
My position is that we want the everything operating in print like conditions. Features such as linear advance require E-steps to be spot on. If we calibrate for conditions different to when we print, we potentially end up with  mismatch and those dependent features not working properly.

PKav89, 2020-08-16:
@teachingtechYT E-Steps/mm is not for "print like condition" calibration, it's just raw axis resolution value, like X-steps/mm or any other axis. Put PLA and you get a value, put ABS - and it will be different. Put flexible - and it will be completely different. Extrusion speed, gear type and tension, temperature, hotend type and nozzle size will also influence calibrated value. Wrong E-Steps value will give you wrong retraction length and wrong unload/load fillament length. For the "print like condition" we have "flow multiplier", which is used only with printing movements and calibrated individually for every fillament.

theely, 2020-08-16:
+1 for E-step calibration to be limited to extruder travel distance calibration, without the influence of external factors.
Material specific adjustments or other factors like hot-end should be taken care by the flow rate adjustment.

E-step calibration is done once and it's specific to the printer hardware.
Flow rate adjustment is done for each filament and is profile based.

One possible approach to limit the number of variables that can influence the E-step calibration is to do the measurement without hot end. Or with a slightly higher than normal temperature.

vector76, 2020-08-16:
I can understand the significance for a bowden setup where loading/unloading needs to match the length of the bowden tube, but for direct drive, it is more of a hassle to measure separately and I'm not seeing the importance of having retraction accurate to within a few percent.

For a direct drive, if the length of travel is calibrated for one filament and includes slippage, then flow for that filament will compensate for diameter.  Then for another filament, flow will compensate for a combination of diameter and any difference in slippage relative to the first filament.  Maybe this is not "correct" in some abstract sense but I'm not seeing it as a practical issue in the direct drive case.

PKav89, 2020-08-16:
@vector76 filament diameter is a different parameter. It exists in any slicer, apart from flow. In order to achieve ideal extrusion, you also need to take some measurements of the fillament diameter and enter average of them into the slicer.

teachingtechYT, 2020-08-17:
The trouble becomes directing the user to disassemble their machine to separate the hot end and give an unobstructed path. For a beginner following the guide, it becomes a point of potential failure as they may lack the experience to correctly reassemble. Once you are knowledgeable in any topic, it can become difficult to remember how daunting these things were when you were a novice. It's a balance I have to keep in mind for any content I make.
Part of the reason I have a slow feedrate for this is to minimise potential resistance.  Do you think a good compromise would be to slow the feedrate again from 10mm/sec to 5mm/sec?
I would argue that if a printer can't reliably/accurately extrude under these conditions then calibrating E-steps is the least of its worries.

theely, 2020-08-17:
Lower feedrate and eventually increased nozzle temperature seems like a reasonable compromise.
I would also consider to document this tradeoff, for expert users to do the proper measurement if they so desire.

teachingtechYT, 2020-08-17:
Agreed. I'll update this portion tomorrow. Thanks for the good discussion everyone.

closed 2020-08-17 by PKav89