teachingtechYT/teachingtechYT.github.io #18: slicer flow calibration doesn't take into consideration horizontal expansion

brainbukt, 2020-08-15:
the slicer flow calibration test doesn't take into consideration horizontal expansion. the method shown for calibrating slicer flow doesn't take this into consideration and can lead to the issue you mention at the bottom of the page re under extrusion it maybe worth checking an tuning this to a degree before adjusting flow as this setting can also reduce elephants foot and bulged corners

vector76, 2020-08-16:
What do you suggest as an improvement?  Perhaps multiple pillars with flat tops and adjustable choices for flow on each of the pillars, similar to the temperature test?

brainbukt, 2020-08-16:
Horizontal expansion setting will actually automatically adjust flow to a degree and is specific to perimeters. It will correct hole sizes and elephants foot I would first tune this setting until foot is gone and then check flow.

teachingtechYT, 2020-08-20:
This one needs a pull request with something concrete. Useful but I'm not sure it would the effort.

closed 2020-08-20 by teachingtechYT