teachingtechYT/teachingtechYT.github.io #22: Temperature Calibration not setting correct temps

69fordf100, 2020-08-15:
The Temp Calibration tower is not changing. I have downloaded the gcode a few times with different temps to see if there would change. In one instance, i started at 195 and for each section, raised the temp 5 degrees stopping at 215. according to the gcode, at layer 51 it goes straight from 195 to 215 and at each new temp section remains at 215. I've included the gcode that I used. I went in and changed the temp changes manually and they worked fine.

M82
M106 S0
M140 S60
M190 S60
M104 S215 T0
M109 S195 T0
G28 ; home all axes
G29 ; probe ABL
;M420 S1 ; restore ABL mesh 
; process Color1-6-1
; layer 1, Z = 0.200

; layer 51, Z = 10.200
M104 S215 T0

; layer 91, Z = 18.200
M104 S215 T0

; layer 131, Z = 26.200
M104 S215 T0

; layer 171, Z = 34.200
M104 S215 T0

OlivierMary, 2020-08-15:
Hi there,

Same for me I set 200,205,210,215,220 and got :

```gcode
M104 S220 T0
M109 S200 T0
...

; layer 51, Z = 10.200
M104 S215 T0
...
; layer 91, Z = 18.200
M104 S220 T0
...
; layer 131, Z = 26.200
M104 S215 T0
...
; layer 171, Z = 34.200
M104 S220 T0
...
M104 S0 ; turn off extruder
```

teachingtechYT, 2020-08-16:
Should be fixed.
There were regular gcode placeholders in the source file: M104 S200
If the user put in a matching video, the javascript was doing a global replace so things could get out of whack and replace more than one.
Now the source gcode has plain English placeholders so only what we want is matched and changed.
Thanks for the testing.

closed 2020-08-16 by teachingtechYT