teachingtechYT/teachingtechYT.github.io #25: Flexbox for navbar

stephen304, 2020-08-15:
Currently the navbar overflows with tabs and looks kinda messy. In order to make the tabs fill the width, I suggest setting display:flex on the container and increasing the mobile navbar breakpoint to 1080px.

Before:
[image]

After:
[image]

lmckeen, 2020-08-16:
https://github.com/teachingtechYT/teachingtechYT.github.io/pull/36

my PR should resolve this issue

teachingtechYT, 2020-08-17:
This one is a matter of personal preference, and my taste prefers the original design. I do appreciate you talking the time to do the pull request.

closed 2020-08-17 by teachingtechYT

lmckeen, 2020-08-17:
sorry @teachingtechYT , the PR for this does not use the new look, but the old one just using flexbox and responsive design, and actually goes much more in to depth with fixing performance issues and a design that works on mobile devices

for desktop devices
before
[image]

after
[image]

and for mobile devices
before
[image]

after
[image]
[image]