teachingtechYT/teachingtechYT.github.io #9: Change calibration types order

zakharsk, 2020-08-15:
In many types of calibration we need temperature and retraction values, but they are in the end of list. May be better do temp/retracts calibrations before use values.

teachingtechYT, 2020-08-15:
Do you mean the order of the tabs on the website?
The things before them I consider as baseline things to get right on the printer, whereas I see retraction, temperature, acceleration, etc as print quality improvements. that's why I put it in this order.

zakharsk, 2020-08-15:
Yes, I meant tabs on the site, because the video and instructions use the term "next step", that is, it is supposed to go through them sequentially.

My proposal came from a personal negative experience, when the printing conditions were incorrectly written on the plastic packaging. It took me two days to figure out what was going on (I'm new to 3D printing).

If your site had appeared two days ago and I had followed the instructions sequentially, then I would not have gone beyond the baseline, because I simply would not have been able to print a cube.

teachingtechYT, 2020-08-15:
I agree frustrating, but in your case the problem was the manufacturers packaging. If you had followed this the baseline cube would have stopped you, which is its job; to identify fundamental problems.

zakharsk, 2020-08-15:
> If there are no major issues, please continue to the next step. If there is a significant defect, the culprit will likely be found by working through the frame page.

Nope )

This is not an important proposal. I just shared my experience so that you have a wide picture.

closed 2020-08-17 by teachingtechYT