teachingtechYT/teachingtechYT.github.io #34: [Feature Request] - Allow setting tuning params to base line print for comparison

theely, 2020-08-16:
For a fair comparison of the newly tuned parameters with the base line cube, the base line cube generator should allow to optionally set the parameters that have been found in the tuning process.
Namely: 
- Flow Rate
- Accelleration
- Linear advance

teachingtechYT, 2020-08-17:
My reasoning is that a new may be overwhelmed by these parameters at this stage. The sole purpose of the baseline print is to see if anything is fundamentally wrong. Introducing these other parameters doesn't enhance this, instead there is more chance of something going wrong.

closed 2020-08-17 by teachingtechYT

theely, 2020-08-17:
What about adding a final stage called: comparison print that allows setting the newly discovered print?
My point is that we should enable users to do a fair comparison with the base line print.
Using the slicer to re-print the cube for comparison will introduce a lot unknown variables.