teachingtechYT/teachingtechYT.github.io #37: [Feature Request] Prime Speed Tuning added to Retraction Tuning

quarky42, 2020-08-16:
I've found that many slicers default to having the same Retraction Prime Speed as the Retraction Speed.  This means for fast retractions the filament is going back into the nozzle area very fast.

Just reducing the prime speed by itself has greatly reduced the stringing I was getting with no further change to the the other Retraction settings.

Please consider adding a prime speed calibration to your Retraction Calibration.   For PETG this helps a lot at reducing the zits and blobs during the reload extrusion move.  I believe Simplify3D has a different name for the setting than Cura does, but it helps in both slicers.  

I have used this setting with both Bowden and direct drive.

vector76, 2020-08-16:
I'm curious how priming speed could affect stringing.  Doesn't priming occur after the movement?  In other words, stop printing one island, retract (possible z hop and wipe), move to new island, prime, begin printing next island.  How would priming affect stringing?

Or is my understanding of the sequence incorrect?

quarky42, 2020-08-16:
An object or fluid in motion tends to continue in motion unless acted in by an equal and opposite force.

The filament loading back into the nozzle too fast, it's going to push more filament out of the nozzle than if you reload the nozzle area a little more slowly.  Not too slow, but when we're talking about a few mm of filament it doesn't take real long to load and carry on.

So if too much filament it pushed out it the nozzle and it is then printing more, you end up building up filament on the nozzle that then gets carried around.  It gets deposited randomly as zits and when it snags somewhere it can get dragged and stringed so the effects will vary with what is being printed.

Tall vertical skinny things / tall edges with open air to one side and a movement to that side will tend to string.  Large flat surfaces will tend to blob and zit.

I've seen this setting help PLA but not anywhere near as much as it helps PETG.

Think of stringing as the end result of an equation of a lot of variables.  You don't know your variable values ahead of time, you just know you have stringing.

So, you eliminate over extrusion, you calibrate retraction (but the slicer automatically makes the reload speed the same as the retract speed but they are two different processes and have two different effects, so this is a bad assumption), you try restart distance to compensate, you calibrate your slicer flow rate so you aren't over extruding the lines, you do all these things and each one holds a component of stringing.

Maybe you get enough variables down to where they need to be the left over error is still absorbed by tolerances on the print or you get only very few issues and it is good enough. Ok.  But, then switch to a new filament and this new one really likes to string, blob, and zit far worse so you need to go further to reduce the erroneous extra bits of filament that are escaping the nozzle at the wrong times.

I hope this gives a framework to look at prime speed as yet another variable.  In some cases, it may not matter much. PLA isn't aa sensitive as PETG to this behavior, but for the filaments it does help it can be wonderful (when used with all the other calibration tools.)

teachingtechYT, 2020-08-17:
Just to be clear, are we talking about the extruder feedrate for the 'unretract' movement just before printing resumes?

quarky42, 2020-08-17:
In Cura it is the Retraction Prime Speed.  It is the speed that the filament returns to the nozzle after a retract.   So yes, it is the unretract speed but I find that term more confusing that way. 

Because the default behavior of this normally hidden setting is to match the retraction speed, it is a major part of how dialing retraction in works, but at higher prime speeds it causes it's own problems.  But since both are tied together by default, you don't get very good control.  

By separating them and testing / controlling them independently, it is possible to maximize retraction speed based on quality without varying the prime speed and then dial in prime speed after that for even better control.   There is a point where more retraction speed causes problems because the prime speed is too high.  Lowering the prime speed allows for a wider range of retraction speed, if that is a better way to explain it.

Having used both direct drive and Bowden printers it seems this makes more of a difference on Bowden, but that doesn't mean it can't help a direct drive.  I've seen an improvement in both when printing PETG, with a bigger improvement on the Bowden side.

quarky42, 2020-08-18:
I believe a better phrasing of this has come to me.  I hope it helps, but I do realize that it has turned into another long explanation.   This explanation is going to hit on a number of things that many people already know, but it is my attempt to try and place Prime Speed in context of these other settings, so I hope you will bear with me here because to me, it seems clear how many of these variables contribute to stringing, so this is my attempt to put it all in perspective.  

First and most importantly:  I look at stringing as an additive or subtractive process.   You can make changes to settings that increase the amount of stringing or increase the likelihood of stringing, or both.   Further more, I look at stringing, blobs, and zits ALL as part of the same physical effect:  "Too Much Filament In the Wrong Spots".   So, when I say stringing, I really mean all unwanted filament deposits.  However, it is critical that everyone understands that just because these are all examples of the same physical effect, the cause of this effect are multiple, varied, and additive.  So, if you can address enough of the causes you can often make your printer print in a way that is unlikely to string/blob/zit most of the time.  Think of it like chipping away at the causes. Once you chip away at it enough, you're left with pretty darn good print quality.  "Perfect" takes even longer to achieve, but "good enough for daily use" is often far easier and more satisfying!

There are a bunch of settings (variables) that contribute to stringing and if you change enough of them you can drastically reduce or nearly eliminate stringing.   That doesn't mean that all potential sources of stringing are eliminated, just that you have lowered the likelihood of stringing down far enough to make it rare / not noticeable.  If you can create conditions where stringing still happens then you can either choose to avoid those conditions or you can choose to try and find further areas where stringing may be reduced.  

 In fact some of the trade-offs involved will force you to have a higher likelihood of stringing because of other effects you are trying to achieve.  So, in order to get to "Zero Stringing" you might have to sacrifice one of the other variables that might not always be possible depending on the purpose of what you are printing.  No one ever said 3d printing was going to be easy.  Okay, lots of people said that 3d printing was easy, but they were either lying or had never 3d printed before.  :)

Some of the causes of "too much filament in the wrong spots" include:
**Overextrusion due to E-Steps / Slicer Flow Rate Calibrations** - In order for your printer to not automatically spit out too much filament at various times / various printing conditions, the printer has to know how many e-steps to generate to spit out 1mm of filament.  If your printer thinks 1mm of filament is X e-steps, but you actually get 1.08mm of filament extruded, that is a lot of extra filament that has to go somewhere / be deposited as blobs/zits/strings depending on how the print head is moving and depending on how the extra filament comes loose or gets snagged.  Put it this way:  1.08mm of filament when 1mm is called for doesn't sound like much, but add that up over 50 or 100mm of filament.   That is 8mm of extra filament that is deposited somewhere on the model when the actual print should have used 100mm of filament.   8mm of blobs, strings, junk... yuck!

But that was just e-steps.  The slicer's flow rate needs to be calibrated as well.  Even if the printer knows how much filament to push out to get 1mm exactly (or even a very slight underextrusion) the printer is dumb.  It ONLY does what it is told and the slicer calculates how much filament to spit out to make model you are printing, so getting the slicer flow rate in line with "reality" is very important. 

**Slow travels** - Gives too time for filament to ooze.  If you slow travels down enough, you can almost always force stringing, zits, and blobs to happen.  Ooze is a form of overextrusion that is time based.  Plus, if filament oozes out of the melt zone in one bad location, that could mean that there is less filament available to start printing at the new travel location once it gets there so you get a combination of overextrusion (strings,blobs,zits left in one location) that ALSO causes slight underextrusion at the new location where the nozzle was going.   

**High temperature** - Filament is being heated hotter than is necessary for good layer adhesion  (This is a tricky one because you are trying to balance physical strength and durability with aesthetics.  In some projects the aesthetics don't matter, only the strength, others still you have to find a balance, and in some projects you just need it to hold itself together nicely, and be as beautiful as possible with strength not mattering at all as long as you have minimal good layer adhesion.)  Again this form of overextrusion can certainly contribute to under extrusion if the melt zone of the nozzle has lost filament.  The first extrusion move after this has happened will be slightly underextruded, but the rest of the print move / extrusion move after is has caught back up might be fine or at least closer to fine.

**Retraction** - Retract too slowly and filament has time to leak out that might not have leaked if you retracted a little faster.  Retract too fast and the filament pulls away from the softer filament in the melt zone an [... cut ...] tion towers.   I really hope that TeachingTechYT who has helped so many people with his videos, will do some experimentation with the prime speed and ultimately consider making a prime speed variable column so that we can try varying that as well.   Default the prime speed to 15mm/s (if you want to use a good medium number) so that when the other variables are dialed in, it is done so without varying this one.  Once those are done, this variable could be changed across a range from 10 to 30mm/s.

drmulligan, 2020-08-18:
If possible a different test model which isolates and demonstrates the effects of prime length and speed would be ideal.

quarky42, 2020-08-18:
The effects of prime speed don't exist in isolation.  I do not believe a dedicated model is necessary, but rather a stringing torture test that is already being used would be a good place to start.

Maybe once tested more, someone even smarter than me could figure out something that might highlight prime speed further.  My initial theory is that multiple tower models are already a good place to start, but the setting for prime speed would need to be limited to 15mm/s during the retraction testing and if someone was inclined to test further, they could then repeat the same tests with variations to the prime speed.

There will be some printers or filament combinations where prime speed may not be contributing enough to really matter.  I didn't notice much prime speed effect when printing PETG Carbon Fiber or Nylon Carbon Fiber.  Both were very thick and not prone to oozing at the temperatures I was printing.  

I like the idea of a least starting with a well known model, do some testing, and then see if any of the results warrant a deeper look or not.  My initial feeling is that just setting the prime speed at a reasonable level and doing the retraction testing will in itself have a significant benefit.  The prime speed would need to be copied over to the slicer at that point.  If someone wanted to go the next step and adjust prime speed up or down, that could be valuable for some.

teachingtechYT, 2020-08-19:
Thanks for the good discussion. Prompted me to add a note about travel speed on the retraction page.
Prime speed is close, it will be implemented in the coming days.

teachingtechYT, 2020-08-20:
Implemented!

closed 2020-08-20 by teachingtechYT