teachingtechYT/teachingtechYT.github.io #46: No flow parameter input for temperature tuning

samroesch, 2020-08-16:
Once a good flow parameter is found, there is no way to use this parameter when printing the retraction tuning or temperature tuning prints

teachingtechYT, 2020-08-17:
It's because the flow is included in the E value for every G1 move. Changing it would involve taking this variable line by line and multiplying by the user input. However, due to slicer variations there's no guarantee the flow calibration would still translate to the tower test prints.
I acknowledge it's not ideal but the tests should be valuable at determining their specific parameters regardless.

closed 2020-08-17 by teachingtechYT

samroesch, 2020-08-17:
Figured at much, still I would have liked to bump up the extrusion multiplier a bit to make my tower print closer to reality, so a field that multiplies the E value would be helpful. Worth noting that a similar tool, the marlin LA gcode generator, does provide this as an option (https://marlinfw.org/tools/lin_advance/k-factor.html)