teachingtechYT/teachingtechYT.github.io #48: G-code generation for delta printers is broken - Includes fix suggestion

thomasloven, 2020-08-16:
Due to the introduction of negative numbers, the G-code generation for delta printers is very much broken.

Ex: Calibration cube for delta printer:
[image]

The problem is caused by decimals being ignored when offsetting the print.
E.g. one of the first G1 commands in baseline.js:
`G1 X37.985 Y35.136 E0.1737 F1800`
This is given an x and y offset of -50, which should be `G1 X-12.015 Y-14.864 E0.1737 F1800`.
However, only the integer part of the values are used in the calculation, and then the old decimals are just pasted back on, which results in `G1 X-13.985 Y-15.136 E0.1737 F1800`.

I believe (but have not confirmed) that a fix would be to change the regex [here](https://github.com/teachingtechYT/teachingtechYT.github.io/blob/master/js/gcodeprocessing.js#L132) to  `/X\d+.?\d*/` and to use `parseFloat` rather than `parseInt` when calculating the new value. The same will have to be done for the Y-offset, of course.

thomasloven, 2020-08-16:
Just saw this is fixed by #43.

teachingtechYT, 2020-08-17:
Have merged the linked pull request.

closed 2020-08-17 by teachingtechYT