teachingtechYT/teachingtechYT.github.io #50: [Feature Request] Gcode scripts support

LordBBQ, 2020-08-17:
It would be very useful if there was support for start/end Gcode scripts. This would also allow for multi extrusion machines as you could insert a 'Tx' command and whatever script your machine uses. Also, this could help for purge buckets, prime lines, and nozzle wiping. This could allow for calibrating parameters like retraction with dual extruder machines, where the Hotends are marginally different, or with different extruders/filaments, ect.

teachingtechYT, 2020-08-17:
I think most people wouldn't need it but it wouldn't be hard to implement so I'm happy to put it in at some stage.

teachingtechYT, 2020-08-18:
Custom start gcode added. Is the end gcode essential? If so, can you give an example.

teachingtechYT, 2020-08-20:
Closing for now unless someone has a compelling example of necessary end gcode.

closed 2020-08-20 by teachingtechYT