teachingtechYT/teachingtechYT.github.io #55: Extrusion Multiplier in the Subsequent Calculators

LordMelkor, 2020-08-17:
I have to set my extrusion multiplier for my current role of filament to 1.06 to get accurate results, however none of the subsequent G-Code generators have anything that allow me to take that into account, so for the retraction tests, if I use the websites GCode generator, it will be generated using under-extruded filament, which I feel like might throw off the retraction calibration.

My only option is to slice the models myself, which is tedious and loses one of the big benefits of the site. :)

Can you include a parameter for Extrusion Multiplier to each of the subsequent calculators?

Printer: Prusa MK3S

vector76, 2020-08-17:
You can achieve this manually by inserting M221 S106 into the gcode.  In principle nothing would prevent the form and the scripts from doing this in the gcode generation, but there may be a balance between option proliferation and aiming to be beginner-friendly.  My opinion is I would agree, especially since it's one of the parameters determined earlier.

I could implement it if Michael agrees that it's desirable.

teachingtechYT, 2020-08-18:
That is a good implementation. The only trouble is the difference in slicers. The original profile used had 0.9 flow rate, so if a user desires 0.9 as well, it will be 0.9*0.9=0.81
Any thoughts?

LordMelkor, 2020-08-18:
> You can achieve this manually by inserting M221 S106 into the gcode. In principle nothing would prevent the form and the scripts from doing this in the gcode generation, but there may be a balance [...]

Is M221 the same as Extrusion Multiplier in Slic3er/PrusaSlicer, though?

Prusa's default printer startcode has the following logic:
M221 S{if layer_height<0.075}100{else}95{endif}

When I look at the actual gcode for something I slice with the default printer start-code + a extrusion multiplier set to 1.06, I don't see a follow up explicitly resetting it to 106. It's incorporating it in some other way.

Is that what you're referring to @teachingtechYT when you say "The only trouble is the difference in slicers."?

teachingtechYT, 2020-08-18:
I mean that although the concept of the flow rate being a multiplier for the G1 E value is the same, the actual implementation behind the scenes may differ. So that 0.9 in one does not match 90% in another.

vector76, 2020-08-18:
Hmm that's a problem.  I assume that is why for the flow page the instructions say to slice your own.  If you had pre-sliced gcode and used M221 to dial in a factor, then when that number is entered into your slicer it could modify it again depending on the other settings, horizontal expansion, etc.

The alternative is if the flow adjustment were "relative to Michael's slicer" and then at the very end the flow could be fine tuned based on the user's slicer.  I'm imagining a scenario where the user accumulates settings from each step to the next, and "flow relative to Michael's slicer" would work for subsequent steps even if it was a bit off for the user's slicer.

It kinda sucks.  I'm not really liking it, but that's the best I could come up with.

teachingtechYT, 2020-08-27:
As discussed, this one can't be implemented accurately. To assist users, the following has been added to the bottom of the flow rate page:

> Can I use this flow value in the other tests on this site?
> The short answer is: not really.
> 
> The gcode generators on this site work by using javacsript to modify source gcode originally created by Simplify3D. However, when you completed the calibration test above, you sliced your own gcode, making your own baseline and then making a flow adjustment relative to that. Therefore, this test is unique from the others on this site which is why the flow rate doesn't necessarily translate.
> 
> Let's say your old flow rate was 100% and you have tested and corrected this to 96%. The gcode on this site originally had a flow rate of 90% when sliced, so applying your 96% to that gives a final result of 86.4%, not 96%. Your slicer profile settings will also be different in other ways, which further complicates matters. Therefore, there is not a straightforward correlation between your slicer and my gcode generators.
> 
> The aim of the site is to discover ideal settings you can apply to your own slicer profile, not to optimise the gcode created by the generators. Keep this in mind and focus on the aim of each test, rather than the general print quality.
> 
> If you are experiencing significant over or under extrusion that prevents you from using the tests properly, by using the custom start gcode function on this site you can optionally issue an M221 to override the values in the generatored gcode. For example, using M221 S90 would tell the firmware to only extrude 90% of what the gcode asks for. This is an easy method for making a quick correction that will alow the tests to complete successfully.

Hopefully this makes the situation clearer. The custom start gcode section also now expands to have more instructions for using M221 and a reference to this new text.

closed 2020-08-27 by teachingtechYT