teachingtechYT/teachingtechYT.github.io #56: [Feature Request] Ironing Calibration

phoob-stack, 2020-08-17:
A Nice feature would be a ironing calibration tool, Ironing is used to get a smooth finish on top of the print. Its quite hard to find a good setting so a ironing calibration tool would be nice to figure out which settings are the best. instead of doing multiple prints, print one simple stair case and on each level it have different ironing settings.

teachingtechYT, 2020-08-19:
As a S3D user I'm aware of this feature and agree it is useful, although I'm not at all familiar with how the gcode looks when it is working. I'm guessing a series of travel moves at the end?
What input would you vary on a stair case for each segment?

teachingtechYT, 2021-07-15:
After considering it for some time, I don't think I can support this due to the limitations of S3D. If they ever update it and include ironing I am happy to revisit.

closed 2021-07-15 by teachingtechYT