teachingtechYT/teachingtechYT.github.io #57: Slicer flow calibration doesn't take into account perimeter overlap

physical256, 2020-08-18:
I would suggest using two perimeters for the calibration of flow rather than a single perimeter.

Guizz27, 2020-08-18:
i think using multiple perimeters introduces too many variables. 
- the slicer may compensate for some overlap
- the printers X/Y precision may have the second perimeter slightly off so the 2 perimeters maybe be slightly closer or more apart
- layer height could have an impact since with higher high, there is more room when the material could go to fill the empty space. Imagine your flow is too high, at 0.3 layer height, the extra extruded plastic could go into empty spaces and you would still get a perfect 0.8 wall thickness but then if you test at 0.1 layer height your wall could be thicker because that extra material has nowhere to go.

I think it's the slicer's job to compensate for overlap and empty space filling.

teachingtechYT, 2020-08-19:
I agree with Guizz27. I see the merits in including a two perimeter test, but it does introduce extra variables and complexity for a beginner.

A special note can be added if you would like to suggest the wording?

teachingtechYT, 2020-08-19:
I've added a note about this near the top of the page. Since the user is slicing themselves, they have complete control over how they slice the cube.

closed 2020-08-19 by teachingtechYT