teachingtechYT/teachingtechYT.github.io #58: Checking 'additional start gcode' in baseline print causes no gcode to be generated

IronMonkeyL255, 2020-08-18:
Checking 'additional start gcode' in baseline print causes no gcode to be generated when pressing the button at the bottom of the page. I changed the bed size (250x210), ABL method (Prusa MK3) and retraction distance (1mm) and it generated the code fine, but it would not generate when I attempted to add a purge line at the beginning. I even tried it with the box checked and no code in there, and it generated fine after unchecking the box.

vector76, 2020-08-18:
I think there could be a bug depending what code is entered.  I put a dummy command G123 XYZ and it choked trying to do coordinate manipulation on an X coordinate of "YZ".  What exactly did you enter for your start code?

IronMonkeyL255, 2020-08-18:
I copy pasted from Prusaslicer, so there shouldn't be any issues with it.

;go outside print area
G1 Y-3.0 F1000.0
G1 Z0.4 F1000.0
; select extruder
Tc
; purge line
G1 X55.0 F2000.0
G1 Z0.3 F1000.0
G92 E0.0
G1 X240.0 E25.0 F2200.0
G1 Y-2.0 F1000.0
G1 X55.0 E25 F1400.0
G1 Z0.20 F1000.0
G1 X5.0 E4.0 F1000.0

It also didn't generate when there wasn't anything in the box.

I was able to get it in there manually, and it ran fine (except it needed a bit more purging because of the MMU2S). I also had to add the Tx tool select prompt.

IronMonkeyL255, 2020-08-18:
I just tested it in the 'retraction tuning' section, and checking the box still allows it to generate the gcode, but as soon as I have something in the box it fails until I clear the box..

teachingtechYT, 2020-08-18:
Thank you @vector76

PR #59 (2020-08-18): Substitute custom start code last, so it's not subject to later substitution
This should fix #58 where start gcode can cause generation to fail.

Not only can it cause failures, it could succeed at producing wrong code, e.g. G92 X0 Y0 could be modified on a delta printer to something undesirable.

By subsituting ;customstart after everything else, the users's gcode should remain intact.
closed 2020-08-18 by teachingtechYT