teachingtechYT/teachingtechYT.github.io #63: [Feature Request] Additional cooling fan options

cheronobyl, 2020-08-18:
It would be helpful to have more options for cooling fan settings in all of the gcode generators. Being able to set a percentage @ a layer would assist people with less standard build surfaces.

teachingtechYT, 2020-08-20:
This is useful although I'd still prefer some presets to preserve sanity in implementation.
I assume this is mainly for PLA and filaments where the fan will be on the entire print?

cheronobyl, 2020-08-20:
Yes, my specific use case is to have the fan come on at later levels when printing with PLA on my PEX-equipped WhamBam. Their literature suggests 3-4 layers of no fan speed and I'm having issues only having it off during the first layer.

teachingtechYT, 2020-08-20:
Ok I'll try and get this one in over the next few days.