teachingtechYT/teachingtechYT.github.io #64: Temperature tower needs print speed field

Spudflyer1, 2020-08-18:
I used the temperature tower with PETG and it printed too fast, would be very helpful to have a field to enter the desired print speed.

teachingtechYT, 2020-08-19:
As a work around could you turn the dial on your LCD to modify the feedrate as you print?

The acceleration test is easy to set the feedrate for because there is only an inner and outer perimeter, so only two speeds. On the other prints, different factors are at play depending on whether the feature is an inner/outer perimeter, infill, solid infill, bridge, travel, etc. It's much more complicated to implement so I prefer the simpler workaround above.

Spudflyer1, 2020-08-19:
Yes I should have tried the dial on the LCD, this will work fine. Thanks for the suggestion! Enjoy.

closed 2020-08-19 by Spudflyer1