teachingtechYT/teachingtechYT.github.io #65: Acceleration Tuning

Asterchades, 2020-08-18:
Just wanted to bring up what appears to be three issues on your Acceleration Tuning page.

Firstly, there's no indication that the model has been deliberately rotated. At least, I believe you've deliberately rotated it. The X is embossed on the side parallel to the Y axis and vice versa, which I'm assuming is to indicate which acceleration/jerk is responsible for the ringing occurring on the print. It's possible you mention this in the video but I didn't find it when scanning through quickly.

Second, the generated GCode isn't 100% compatible with Klipper firmware. While Marlin seems to not mind the missing T parameter for M204, Klipper will completely ignore the command by design. Using the legacy (at least in Marlin) S parameter works fine, as does specifying both P and T (which Klipper will take the lowest of). I'm unsure how RepRap handles the missing parameter as I don't have a Duet-based printer, though I suspect it's more graceful than Klipper.

Lastly, I feel as though a better demonstration picture may be in order - or perhaps at least a little more explanation. While I can see the subtle reduction in the after-ringing that you're talking about as you go down the accelerations, there appears to be more pre-ringing as you go down further as well. I'm unsure whether you consider this a non-issue or if it is your intention that this be solved by way of a different method (jerk/JD tuning? Linear advance?) that isn't mentioned. _Personal opinion, it's a toss-up between 500 and 600 due to how even it is on both sides, where 400 seems to introduce noticeable pre-ringing._

Otherwise (and these are admittedly only minor), major kudos for the effort. The GCode generation has finally inspired me to finish dialing in my settings, mostly because of how much of a nuisance it is to do the same with Cura. Frankly it's easier at times to manually edit the GCode, which is unfortunate given how well the rest of the program tends to work.

teachingtechYT, 2020-08-19:
1. The X is purposely on the Y side and vice versa as I figured a wobble in one axis would show on the other.
2. This was built for Marin but I'm happy to support Klipper and other firmwares. I have tried Klipper but am nowhere near as familiar with it. I figure it would be easy to add Klipper ABL support to the ABL drop downs. Would implementing the P and T require the user to find these values from M503 and input them into the form so the generator could inject accordingly?
3. Perhaps embedding the relevant snippet from my original acceleration test video would be useful here after the sample image? I believe the pre-ringing is actually bulging from the hot end being in the one spot too long.

Asterchades, 2020-08-19:
X/Y labels is what I suspected, then. Makes sense, though perhaps not to a beginner.

M503, unfortunately, isn't supported in Klipper. Nor are Jerk or Junction Deviation, but because they're superfluous and attempting to set them won't matter it's less of an issue. Instead they use a parameter called "Square Corner Velocity" which is functionally very similar to Junction Deviation but uses a larger value (the default is 5mm/s).

To get the settings from Klipper, you need to send "set_velocity_limit". This will list the maximum velocity (default 300mm/s), max acceleration, max acceleration to deceleration (defaults to half the acceleration, but I don't know if M204 updates it), and the square corner velocity, with their respective parameter names. It's convoluted (at least in my opinion, given the existing, documented commands for Marlin) but that's how they want to play it. If you want to adjust the other values you have to address them through their "human readable" GCode: "set_velocity_limit max_velocity=<value> max_accel_to_decel=<value> square_corner_velocity=<value>".

As I mentioned, if you use "M204 S<value>" than it will automatically treat it as though you sent "set_velocity_limit max_accel=<value>". Alternatively you can use "M204 P<value1> T<value2>" and it will again run the same command using whichever is the lower out of value1 and value2.

ABL I can't 100% help with. While I use a BL-Touch, I mostly use it as a replacement for the Z stop so I can switch bed platforms without having to adjust the switch location. It uses the same sort of "human readable" instructions in the form of "bed_mesh_calibrate" to just run it or "bed_mesh_profile load=<name>" to recall saved details.

I'm more than happy to answer whatever questions you may have about Klipper, though I'm not going to know everything as I'm just an end user. In the meantime there might at least be some useful info here:
https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md

Very curious about the pre-ringing, though. The image could be made more sensible by the relevant portion of the video, especially given the not-entirely-related cause.

closed 2020-08-30 by teachingtechYT