teachingtechYT/teachingtechYT.github.io #66: Add GCode comments showing chosen parameters

drmulligan, 2020-08-18:
I changed my chosen parameters on the retraction tuning tab and then thought to label my first test.  I looked inside the GCode to see if I could see what my choices were, they were not there.  It would be helpful if the parameters the user chose are written to the file as comments for later discovery/recovery.
These comments could also be machine parsable, allowing the calibration tool to read them back.

teachingtechYT, 2020-08-18:
Are you talking about adding a comment to the end of the affected line of gcode so it is easy to find them in a text editor to verify the change?
If so that would be very easy to implement:
eg. ; custom retraction
; custom temperature

teachingtechYT, 2020-08-18:
Added for all tests. Search for 'custom'. Segments are labelled A-F where applicable for those tests.

closed 2020-08-18 by teachingtechYT