teachingtechYT/teachingtechYT.github.io #70: [Temp Tower] Printing stops after level A

goombah88, 2020-08-19:
Trying fourth time as I type this. Every time the printer just stops after it starts on level B. The print head is unlocked and no more filament is coming out. It just stops dead. Using a Bibo2 Touch. Tried redoing settings, reexporting from site. Same file size every time and same result.

goombah88, 2020-08-19:
1st 3 times I tried it were 190 and up, in increments of 5. This time I used 185 and up, in increments of 5 and it worked...go figure...bug?

vector76, 2020-08-19:
What other options were you using?  When you say print head unlocked does that mean stepper not engaged and you can move it easily?

goombah88, 2020-08-19:
Center box was checked, retraction of 2/40,and yes, the stepper motors were
disengaged so I could easily slide the print head around.

Thanks,

Chad

vector76, 2020-08-19:
I am not finding anything unusual when I try to recreate the same thing.  When I generate with (190, 195, 200, 205, 210) compared to (185, 190, 195, 200, 205), the only difference in the files is the M104 lines (as it should be) and I can't see how those could produce the behavior you see.

Can you attach a gcode file that is failing and one that is succeeding?  I suspect something else is happening unrelated to the gcode.

goombah88, 2020-08-19:
I kept overriding the G-Code file each time. I will try and recreate the
issue tomorrow and let you know.

Thanks,

Chad

teachingtechYT, 2020-08-19:
I've just updated how it functions, not that I would expect it to matter here, but worth another shot nonetheless.

goombah88, 2020-08-19:
Well, I just refreshed and exported gcode starting at 185 and 190 and both
printed to completion. I'm not sure what happened but it at least seems to
be printing now.

On a side note, however, the 4 prints I've gotten to complete now all look
like crap. The bridges have bad defects at all temps. I just printed a
skull with horns yesterday with the same filament and it turned out
beautifully so I'm a little perplexed. I'll try a different bridging test
from thingiverse to compare the results.

goombah88, 2020-08-19:
Here you can see what I mean. The one on the left starts at 185 and the one
io the right starts at190. Normally I print at 190 with this filament

Thanks,

Chad

teachingtechYT, 2020-08-19:
Is there meant to be a pic?

goombah88, 2020-08-19:
That's odd. I see the pic attached to my email.

goombah88, 2020-08-19:
Thanks,

Chad

goombah88, 2020-08-19:
It's kind of hard to see but the bridging on this test worked great oh, so
I think it must be an issue with the default settings that your gcode uses.
Picture attached.

Thanks,

Chad

vector76, 2020-08-19:
Not seeing pictures, so I think attaching by email is not supported.  You probably have to log in to github to attach them.

Although I think it probably deserves a new issue because this is no longer about stopping after level A.  From what you've said I think it's safe to close this issue.

closed 2020-08-20 by teachingtechYT