teachingtechYT/teachingtechYT.github.io #73: [TEMP TOWER] Bridging code problem

goombah88, 2020-08-19:
Of the 4 prints I've gotten to complete now all look like crap. The bridges have bad defects at all temps. I just printed a skull with horns yesterday with the same filament at 190 and it turned out beautifully so I'm a little perplexed. I tried a different bridging test from thingiverse to compare the results, printed at 190, and it turned out just fine. So I think it's a problem with the code generated by the site.

Here you can see what I mean. The one on the left starts at 185 and the one io the right starts at190. Normally I print at 190 with this filament.
[image]

Here's the other test I ran:
[image]

teachingtechYT, 2020-08-20:
The logic for extruding the bridge is different to a regular overhang. The perimeters will be in mid air and then the solid infill has to bridge between the perimeters with a sufficient amount of overlap to catch/grip on each side. If the flow rate is under (which could just be from my gcode source even if your slicer is set up well), the infill may look like your pic.
Ultimately the test is for temperatures and the result you have shown doesn't stop it being useful. We have to remember this site is a just a tool for tweaking the printer/slicer for your real prints. If you are happy with the skull with horns, that's all that matters.

teachingtechYT, 2020-08-25:
Going to close this one soon pending an update from the OP.

goombah88, 2020-08-25:
I guess go ahead and close it. I'll open a new case if I have any more issues with it since your code change.

closed 2020-08-25 by goombah88

teachingtechYT, 2020-08-27:
It's worth noting that there is now more info on adjusting flow rate if needed in the custom start gcode section.