teachingtechYT/teachingtechYT.github.io #74: Floating Point Rounding Errors

dkatzdev, 2020-08-19:
While looking at gcode commands I have noticed a lot of numbers end with things like .99999999999 or .00000000001. These are the results of rounding errors in the code. While it probably wont effect the tests much, it is indicative that the wrong functions are being used somewhere.

vector76, 2020-08-19:
I agree this is ugly but I don't think it's a latent bug.  I think this is fundamental to floating point numbers and how decimals are not exactly representable.

If you open a javascript console (F12 in most browsers and select "console"), try entering 100.3 - 100.  I get 0.29999999999999716 and there is no wrong function being applied.

It might make sense to truncate the values to 0.0001, because it is still a bit ugly.  But like I said above I don't think it's an indication of a latent bug.

teachingtechYT, 2020-08-20:
I agree it makes the gcode messy but doesn't change how it prints so we can safely ignore.

closed 2020-08-20 by teachingtechYT