teachingtechYT/teachingtechYT.github.io #81: [FR] Firmware retraction support

gayuha, 2020-08-21:
It would be nice to have support for firmware retraction. Maybe a checkbox that enables or disables it.

dennisrosa, 2020-09-03:
Hi,

I create a PR #123 with firmware retraction feature and linear advance.
Unfortunately I only tested in RRF and works fine

Best Regards

teachingtechYT, 2020-09-03:
Sorry I haven't replied to the PL yet. I would like to include it but on a separate page. At the moment there is already five values per segment which is daunting for beginners.
I would propose to implement the changes by adding a second 'advanced' retraction test. More control for those who need it and less on the screen for the average user.

dennisrosa, 2020-09-03:
No Problem !

Maybe a toggle between normal and expert mode.

croaker-1, 2020-10-29:
A more simplistic version of this could be a single checkbox for Firmware Retraction and something like this (as an example):

```
    // A section
    if(fwretract){
        retraction = retraction.replace(/;retractionA/g, "M207 S"+a1+" F"+a2+" Z"+a5+"\r\nG10" ; custom retraction - A");
        retraction = retraction.replace(/;unretractionA/g, "M208 S"+a3+" F"+a4+"\r\nG11" ; custom un-retraction/prime - A");
    }
    else {
        retraction = retraction.replace(/;retractionA/g, "G1 E-"+a1+" F"+a2+" ; custom retraction - A");
        retraction = retraction.replace(/;unretractionA/g, "G1 E"+a3+" F"+a4+" ; custom un-retraction/prime - A");
        if(a5 > 0){
            retraction = retraction.replace(/;zhopupA/g, "G91\nG1 Z"+a5+" F1200 ; custom z hop - A\nG90");
        }
    }
```

I'm no programmer but I could probably make the changes figure out how to do a PR if needed.

teachingtechYT, 2021-07-13:
I have just rewritten the source S3D profile and implemented z hop for all tests. This change should make firmware retraction easier in the future. The retraction page will need a lot of reworking to not show too much information. I think I'd probably like to make a general video on firmware retraction first too.