teachingtechYT/teachingtechYT.github.io #100: Additional G Code (Before ABL and Temp Set)

kiowadriver, 2020-08-23:
Some printers such as mine must be issued the M80 G-code to first turn on the stepper and heater PSU prior to BL and preheat.  The "additional G-code" feature you have applies the g-code after these moves.  Can you place it prior?

teachingtechYT, 2020-08-24:
Most people use the start gcode for an extruder prime/purge so that's why it comes after the rest.
Unfortunately moving it before will be worse for the majority of users. 
It could be added to all of the gcode at the beginning. From what I've noticed, Marlin just ignores codes it doesn't support, so hopefully no harm done. Will test.

teachingtechYT, 2020-08-24:
Tick box added for each generator. M80 is added as the first line of gcode.
I left M81 out of the en gcode because some heat sink fans need to keep running to avoid hot end issues.

closed 2020-08-24 by teachingtechYT