teachingtechYT/teachingtechYT.github.io #101: [Feature Request] Flow and Linear Advance Parameter

sortagreg, 2020-08-23:
## REQUEST
Add an optional flow and linear advance input, on sections where they would be applicable, such as retraction, temperature and acceleration tuning. 

## DESCRIPTION
In the steps past calculating flow(retraction, temperature, and acceleration tuning), there is no option to input the prior calculated flow value into the these steps. Over/under extrusion could affect the final results.

Furthermore, in the notes for calculating retraction, it is mentioned that linear advance may have a severe impact on retraction, but, like with flow, there is no way to input this value, if known, into the form to see and test the changes, without adding the line to the gcode manually, after download.

teachingtechYT, 2020-08-24:
Linear advance K factor can be set with M900 by utilising the custom start gcode feature.
The flow is not so simple, see here: #55 
M221 could be used but it will not be exact due to the differences in the source slicer profile that I've used vs what the user has calculated in their own slicer. Nevertheless I will be adding notes about this so users at least know about the option.

teachingtechYT, 2020-08-27:
I've added some specific notes on why a user's calibrated slicer flow rate doesn't directly translate to these tests. The custom start gcode section now expands to have more specific instructions on implementing the features you are talking about.

closed 2020-08-27 by teachingtechYT