teachingtechYT/teachingtechYT.github.io #98: [Feature Request] Mix tower geometries in retraction.stl to facilitate diagnosing issues from movement computation delay

dmarx, 2020-08-23:
I was performing retraction tuning and observed small random irregularities in tower exteriors which I (at first) attributed to high temperature oozing. I built [this temperature tower](https://www.thingiverse.com/thing:3064519) and noticed the same irregularities in the cyndrical tower even at temperatures where there wasn't stringing, while the square tower had no such irregularities at any temperature. 

I'm pretty sure what's going on is the issue described [here](https://www.reddit.com/r/3Dprinting/comments/dhaudy/poor_print_quality_with_octoprint_any_suggestions/) caused by sending commands from octoprint to the OEM ender 3 pro board via USB, resulting in the nozzle pausing to wait for commands on the more complex circular geometry, resulting in the observed bulges. 

My recommendation is to change the tower shapes in retraction.stl from a pair of cylinders to one cylinderical tower paired with a simpler geometry like a rectangular or triangular tower to facilitate diagnosing problems caused by gcode processing speed.

teachingtechYT, 2020-08-24:
This is a good suggestion. I will put this on the to do list. I'll probably use a triangle with the inner corners pointing inwards.

teachingtechYT, 2021-07-13:
This is fixed in a update this week with pentagon shaped towers. Less gcode to process and quicker to print too.

closed 2021-07-13 by teachingtechYT