teachingtechYT/teachingtechYT.github.io #99: [Feature Request] Acceleration Tuning - First layer with brim

C0unt23r0, 2020-08-23:
First layer is printed always filled which takes increased amount of time. In your linked video description you are telling to disable the first layer and add optionally a brim as support.

I would like to see this directly happen in the generated GCode.

teachingtechYT, 2020-08-24:
It's like this on purpose because of the size of the print. My target for the generator is for everything to fit on a 100 x 100mm bed. I had to shrink the original model to do this and disable a skirt to make sure.
Expanding the first layer horizontally kills compatibility with some printers. Not many, but as an owner of a Monoprice Mini Delta, some.

C0unt23r0, 2020-08-24:
Ah. Well that's understandable and a good point too.

As I am always overcomplicating things you could add a condition to switch between one acceleration template with a solid bottom layer and another template with a brim. Totally overkill as you have to maintain two templates when updating.... I know :)

At least I can say thanks. I am new to 3d printing and it was good to understand and helped me a lot!

teachingtechYT, 2020-08-24:
Thanks for the feedback. I tried to see if I could implement a 'reverse brim' yesterday, so that a full first layer isn't needed. Had no luck I'm afraid.

closed 2020-08-24 by teachingtechYT