teachingtechYT/teachingtechYT.github.io #104: [Feature Request] export retraction settings as slicer profile

jscancella, 2020-08-24:
There are a lot of assumptions built into the stringing/retraction tuning. It would be nice if after you generate  one that produces no stringing you could export it as a Cura (or other slicer) profile. Because just moving over some of the settings doesn't always produce the same results.

Example: https://www.reddit.com/r/ender3/comments/ifqvh5/stringing_issues/

teachingtechYT, 2020-08-24:
If I were to write code that output a slicer profile as you request, it would still be moving over settings. The time required to implement that vs a user copying a few settings is excessive.
I don't know what you mean about assumptions build in. It simple substitutes gcode based on the user's input. It doesn't claim to do anything magic or different to that.
The linked example is noticeably different. I would suggest for that user that they read the related section of the page. My guess is coast and/or wipe being present in the Cura profile. I'm happy to add anything else relevant to that list.
```
Other factors beyond the scope of this test - Important!
Retraction acceleration: This will affect whether the retraction speed can actually be reached. The gcode generator below does not include any changes to what is set on your printer. You can change this with M204 and the R argument.
Slicer settings such as coast and wipe: Coast stops extrusion slightly early to assist retraction. It effectively lets the hot end 'run dry' at the end of the printing movement to reduce ooze. This varies from slicer to slicer and isn't always necessary to tune.
Wipe moves the nozzle back towards the recently printed geometry to wipe ooze off. If you are having trouble reducing stringing, it may be a good option.
Both coast and wipe are turned off in the gcode generator below.
Travel feedrate and acceleration: A travel move is one where the printer moves to a new location without extruding. The slower this move is, the more time filament will have to ooze from the nozzle and add to stringing. The feedrate is set to 100mm/sec in the gcode generator below, and does not include any changes to what is set on your printer for acceleration. You can change travel acceleration with M204 and the T argument.
Linear advance: Linear advance, covered later in this guide, can drastically improve the accuracy of our extrusion. It has a significant impact of retraction (reducing the need), so after configuring linear advance you may need to revisit retraction.
Slicer differences: The gcode generated below was originally sliced by Simplify3D. The settings you establish should translate to your slicer quite well but there may be idiosyncrasies. For instance, Cura measures extra restart distance in volume rather than length.
```

jscancella, 2020-08-24:
> I don't know what you mean about assumptions built in. 
By that I mean things like nozzle size, layer height, etc. 

>The time required to implement that vs a user copying a few settings is excessive.
If you don't want to export profiles, there needs to be a way to see why your gcode is producing good results but moving over the described settings doesn't yet the same results. Ultimately if a user isn't able to replicate the results like they can when using your gcode then the tool isn't helpful. Perhaps instead you could provide a default profile for each slicer so that the user is starting from the same point you are when modifying the gcode?

teachingtechYT, 2020-08-24:
The nozzle size, layer height, slicer used are all listed on the first tab of the site under a giant "Warning - Read carefully!" heading.
As I quoted above, there is already a list of things to try on the retraction page to explain discrepancies. What you are requesting is already there. If someone chooses to ignore this, it is not fair to blame the site and say the "tool isn't useful".
I can't supply baseline profiles for slicers I don't use, especially since they won't be relevant to the gcode generated on this site, originally by Simplify3D.
You are asking me to triple my workload so a user doesn't have to copy over settings or read the page. This is a completely free and open source site. This seems very unreasonable to me.

jscancella, 2020-08-25:
I'm not trying to increase your work, just come up with a solution. It's your website, if you don't want to change it that's up to you. However I think it is harsh for you to say I am being unreasonable. I listed this as a feature request not a bug, and am open to other solutions. Since you don't seem interested in my comment I will close this request.

closed 2020-08-25 by jscancella

teachingtechYT, 2020-08-25:
For me it's pretty simple. You need to read the site. You said there were assumptions, when the details are listed on the first tab. You asked for steps to get the retraction to match, they are already there.

Exporting a slicing profile is not viable regardless. They have hundreds of parameters. Here we test for a dozen or so. If I output a slicer profile, defaults would be used for everything besides the ones from the test, overwriting any already customised by the user, unless the user wants to manually input all the missing ones. So it ends up being more work and less efficient for everyone.

If you look at the update history, the site has undergone tremendous improvement thanks to bug fixes and new features. I am thankful for feedback and suggestions. They just need to come after the user has used the site as intended, not ignored the instructions.