teachingtechYT/teachingtechYT.github.io #105: Retraction Tuning

teapowell, 2020-08-24:
Not entirely sure what is going on here but when I use the gcode created by the website I get very good results but when I put those same settings in a default 0.2 layer height Cura profile I get really bad stringing.

In my case the retraction settings I optimised to were:
* Retraction Distance = 3.3mm
* Retraction Speed = 50mm/s

I am using an Ender 3 Pro stock (except dual gear metal extruder). Printing PLA at 200/60 & 50mm/s print speed.

Reddit post with pictures: https://www.reddit.com/r/FixMyPrint/comments/ifow25/stringing_issues/ 

Why are the results so drastically different?

teachingtechYT, 2020-08-24:
Someone ninja beat you to posting your link. Please see my reply here: #104

teachingtechYT, 2020-08-25:
The other issue has gotten off track, so I'll post here.
Please see the 'Other factors beyond the scope of this test - Important!' notes on the retraction page.
There is something set differently between your Cura profile and the gcode generated on the site (originally from S3D).
The most obvious ones to check are coast and wipe, which are both turned off for me but may be turned on in Cura.
Work through the list and get each parameter to match.

teapowell, 2020-08-25:
Thanks for responding.

Firstly, I would like to address why I opened this issue (because of the other issue thread). I am finding your site very helpful and informative, I opened this issue as I believe Cura is one of the most popular slicers (correct me if I am wrong) and I transferred over the settings into a default 0.2mm profile, which I believe many people will do. With the results being very different, I either did something very wrong or I missed a step in translating the settings across.

I have checked my profile and matched all the settings and still get the same result. I do have Z hop off, but the gcode your site generates has 0mm for the Z hop, is this the same as off? Or could it be due to the Retraction Minimum Travel? By default that is set to 1.5mm, I am wondering if retraction is even occuring on this test?

teachingtechYT, 2020-08-25:
Agreed Cura is the most popular slicer. Just remember that this site in not Ender 3 specific, even though it's a popular 3D printer so many people will be tuning this combo.
Z hop of zero is the same as having it off. There are comments in place for z hop that stay as comments (and therefore ignored by printer) if you leave it at zero.
Interesting thought on the retraction minimum travel. Since the end profiles are round, they are effectively made up many, tiny straight lines. Could you post the gcode output by Cura using the settings mentioned above by using pastebin.com or filebin.net ?
Then I can check if the retraction commands are in place.

teapowell, 2020-08-26:
Here you go: https://filebin.net/40y3v5t57w0ql6nh

I forgot to say that I had tuned the flow. Other than that it is a default 0.2mm Cura profile.

teachingtechYT, 2020-08-26:
This is pretty interesting. This is a preview of the gcode made by the site:
[image]
Your file (and it looks the same to me when I slice in Cura:
[image]
The S3D gcode I would describe as elegant. The Cura gcode is a total mess. So many travel moves. From what I can tell, it retracts on one side, then does a little zig zag before the main travel move to the other side, another zig zag before unretracting and printing again.
I'm not familiar enough to know what setting would be making the travel and retraction so messy but no doubt it's the reason behind the stringing. The retraction values in the gcode are what you've entered but the logic of the toolpath around them is very erratic.

teachingtechYT, 2020-08-26:
Here is what is happening on each layer. Cyan is infill, dark blue is shell, green is travel moves, red and blue circles are retracts and primes.
[image]

teapowell, 2020-08-26:
Well that is really strange! I just had a play with settings in Cura and viewing the travel lines, when I set the infill to 0% I get the straight sensible travels, when it is anything else the travel is erratic. Could it be trying to do all the infill first before the shell?

Thanks for deep diving into this with me.

teapowell, 2020-08-26:
Also, by default Combing = Not In Skin in Cura. Could this contribute?

teapowell, 2020-08-26:
Playing around with some settings and viewing them in Cura, it seems that Infill, Z Seam Alignment, and Combing all affect the path. I have uploaded a new gcode to the Filebin. This one has Infill at 0% all other options are default (except the retract distance and speed which I set to 3.3 and 50 from testing with your gcode).

teachingtechYT, 2020-08-26:
This is what your new file looks like for me which is a huge improvement:
[image]
How does it print?
I'm also interested to see how one of your regular prints performs with the calibrated settings but not these new changes. A more natural test.

teapowell, 2020-08-27:
[image]
So I spent a lot of yesterday doing lots of retraction test printing to see if I could replicate the results from the gcode generated by your website. The attached picture is the best I could get.

Regular prints have a fair amount of stringing and blobbing on, which is on of the reasons I am trying so hard to get the retraction right!

teachingtechYT, 2020-08-27:
Thanks for your time here and the annotated pic. Part of me thinks that that maybe there is an anomaly in your printer such as degraded bowden tube or it not quite being seated correctly, but then you had good results with the 3.3 distance from this site.
I have a thread going in my Patreon forum and we are discussing possible settings.
Just confirming you have coast and wipe both disabled in your profile?
Can you also confirm what your combing is set to?

teapowell, 2020-08-28:
Both coast and wipe are disabled. Combing is the default which is 'Not in Skin'. It is very strange! When I got the results from your print I was very excited that I could achieve no stringing.

teachingtechYT, 2020-08-28:
Try combing off and see what that does. In skin still does the zigzag as far as I know.

If your printer did it, then it is possible. We just need to find the culprit for what's messing it up.

teapowell, 2020-09-01:
Here are the results. I am printing other objects at 10mm retraction and that is giving me passable results but there is still a little bit of stringing and zits.
[image]

(appologies for the late reply, I went away for a few days)

teachingtechYT, 2020-09-01:
I'm really perplexed by this. Thanks for the update. I might do some site vs Cura testing myself on the Ender 3 V2. Your settings are huge btw. 10mm @ 100 mm/sec! If you hadn't gotten the clean 3.3 mm test I'd say something was up with your printer.

teapowell, 2020-09-02:
I'm pleased I am not the only perplexed one! Yes I realise they are huge, I didn't push past those values and was still getting some stringing. It's not the filament either as I have swapped it out for another different brand. Please keep me updated on your tests :)

endobengo, 2020-09-18:
Did you guys ever reach a solution on this?  I'm having the exact same issue.  Using an Ender 5 Pro, site test prints great, Cura and Creality sliced tests do not with same settings.  
[image]

endobengo, 2020-09-19:
Forgive me if I'm being annoying, but I think the issue is more in the extrusion speed.  Comparing the two I find that it drops dramatically in the upright sections under Cura, but no in the site test.  The best I've been able to tweak the settings is to get 24mm/s extrude speed in Cura(increasing or decreasing wall print speed has had wildly erratic results), while the test stays constant at 45.0mm/s.  
[image]

endobengo, 2020-09-19:
Promise this'll be my last post on this, as I've somewhat solved my issue.  They extrude speed was being reduced due to the cooling fan's minimum layer time setting, which in Cura 4.7.1 is default at 10 seconds(way longer than needed to print this).  Creality's layer time was at 5 seconds.  Reduced both to two seconds and am getting consistent speeds now.  Still getting some SLIGHT stringing(possibly due to the erratic travel pattern) but it's WAY better than it was.

khaaaannnn, 2020-11-12:
I'm having a very similar issue where I put the settings from the calibration tool into a stock profile on cura and get different results, I've tried different tweaks as well now to try and get rid of the last little strings on a model and get them looking great with your tool but as soon as I put it into cura I get spiderwebs everywhere again. Will attach photos soon, what infill does your test tool print with? I've disabled zhop, z-seam infill, combing, wiping and coasting to match the more stock settings all while trying to see if maybe one of these settings is what's the key to my specific scenario. Sorry about grammar and formatting I'm on a phone.

spit40, 2020-12-16:
> Promise this'll be my last post on this, as I've somewhat solved my issue. They extrude speed was being reduced due to the cooling fan's minimum layer time setting, which in Cura 4.7.1 is default at [...]

Wow, this is a big find! Thank you. I'll test myself but I think its a big factor in why the gcode generator tests are coming out so much better. Also why a print takes 50 minutes to print when done in Cura and 20 mins from the Gcode generator. With a smaller minimum layer time I get it down to 37 mins. Something else going on I guess to account for this extra 17 mins

teachingtechYT, 2021-07-15:
I really appreciate this robust discussion and testing. I have added to the retraction page to reflect these possible variances so that they may help others in future.

closed 2021-07-15 by teachingtechYT