teachingtechYT/teachingtechYT.github.io #110: What to look for in Junction deviation?

PigeonFX, 2020-08-26:
The video and text say the changes in Junction deviation will be subtle, but subtle how? with subtly reduced rining?

There is a good photo showing the acceleration effects on the Y, but nothing giving a representation of what to look for when comparing the Junction deviation values.

teachingtechYT, 2020-08-26:
When I tested JD, I couldn't see much difference. Perhaps I needed a more extreme range of values.
I think it's one of those things where unless your printer is running with a really high feedrate/acceleration, it's hardly worth tuning.

PigeonFX, 2020-08-26:
Thanks for the info, did you run with Jerk too, and if so any improvements there, know there were a lot of bug reports about JD in 2.0.x wonder if it's still a problem? 

Thanks for all the work you put into this calibration tool, lovely to have a one-stop-shop for all the key tuning factors!

teachingtechYT, 2020-08-26:
I haven't tuned for jerk as the printers I've used to test only ran JD.
I encountered a bug myself back when JS was made the default but I think that has been patched a long time ago. No problems since for me.
Thanks for the nice feedback.

PigeonFX, 2020-08-26:
Maybe as JD is calculated as JD=0.4*Jerk²/acceleration having a Box to put your chosen acceleration into that would then spit out JD values equivalent of 4 to 9 Jerk would be useful over just arbitrary 0.01-0.06?

teachingtechYT, 2020-08-30:
The idea is people use M503 to find their current value and then change the form to choose values either side of this.

teachingtechYT, 2021-07-13:
I am adding an embedded 'interpreting results' video to each section this week and will try to cover this.

closed 2021-07-13 by teachingtechYT