teachingtechYT/teachingtechYT.github.io #111: Feature Request :  Localization - Enable localization

devzwf, 2020-08-26:
Hi,

Would be a great idea to enable some sort of Localization so , we can contribute with translation easier.

Thanks

teachingtechYT, 2020-08-26:
I've never done this before so what exactly would I need to setup for this?

carlosframirezb, 2020-09-20:
I' would like to help with spanish translation! Just let me know...

devzwf, 2021-05-25:
I can contribute with french

teachingtechYT, 2021-07-13:
I am in the middle of a big rewrite, so after that I can come back to this.
phckopper's solution sounds quite workable.
My only concern is that this will require an ongoing commitment to keep the translations valid. Each translated version would need a warning at the top of the page saying it might not be to date with the English version.