teachingtechYT/teachingtechYT.github.io #112: [Feature request] Different filament diameters

Millimeter94, 2020-08-27:
Hi

My printer is using 2.85mm filament and there seems to be no way of telling that to your gcode-generator. Could you add a way to change the filament diameter in the gcode-generator?

Thanks

teachingtechYT, 2020-08-27:
The gcode is already hard coded for 1.75mm, but I wonder if the M221 command could be used to up the percentage to suit?
Click on the custom start gcode box and more instructions will appear.
From 1.75 to 2.85 should be 163% flow: M221 S163
Maybe it's not that simple but if you'd like to be a Guinea pig and test that would be great.

Millimeter94, 2020-08-27:
Reports from the Guinea pig ;)
At least for the first layer test the M221 command seems to work. There was just a small mistake in your calculation. The correct flow rate is 61.4%. I will test the rest of your website tomorrow since it is already 2 am.

teachingtechYT, 2020-08-28:
I was confused on whether to multiply or divide and I guess I got it backwards.
If you can continue verify, I'll add notes with specific instructions on how to replicate.

Millimeter94, 2020-08-28:
Sounds good.

teachingtechYT, 2020-09-01:
Any further testing for this?
I would love to officially add a note for users such as yourself.

Millimeter94, 2020-09-01:
Sorry. I have been a bit busy this week and didn't have the time to run through the full tests. I might get to it on Thursday.

Millimeter94, 2020-09-02:
I was able to do some more tests today. As far as I can tell the gcodes work well when using the M221 command. Sadly my printer seems to be in a pretty bad shape. Especially bridges and small details are extremely difficult to print. So I am not sure how applicable those results are for other people. But I would definitively call it "Works for me". Maybe someone else with a printer that uses 2.85mm filament could try this out as well to see if it also works on printers that are in better shape than mine.

teachingtechYT, 2020-09-03:
Thanks very much for following up. I will add a note as a potential but not proven solution.

teachingtechYT, 2020-09-03:
Notes added on front warning page and for each generator form.

WillWill56, 2020-09-03:
I recently discovered this tuning guide and experimented with using flow rate to adapt the generated gcode to the 3.00mm filament in my printer, so I'll share my findings...

Looking purely at the maths, which involves comparing the cross-sectional area of both 1.75mm and 3.00mm filament, I got a theoretical flow rate of 34.1%, however when printing the baseline cube with the extra "M221 S34" it seemed there was some under-extrusion. In hindsight this might've been poorly adjusted bed-height and a bad first layer, but I ended up increasing the flow rate to 42% and managed to get a successful print. The cube is slightly oversized at 20.30mm in X and Y directions so it's possible that value is in the realm of over-extrusion. I did calibrate my E-steps before printing the baseline so I'm fairly sure I can rule that out as a source of error, and now that I've got my bed height calibrated a little better I could do some more testing.

Following the same thought process, the flow rate for 2.85mm filament with 1.75mm gcode calculates out to 37.8%, but it's possible I'm not using the right maths for this problem...

Millimeter94, 2020-09-03:
Now i'm feeling slightly stupid...
You are completely right. The flow rate has to be compared based on the area and not on the diameter. I just printed the baseline cube again using M221 S38 and it looks much better. Maybe I should stop testing these things in the middle of the night ;)

teachingtechYT, 2020-09-03:
Thanks guys for following up. I also compared diameter instead of area.
I will update the text on the site.

teachingtechYT, 2020-09-03:
Done and thank you both.

closed 2020-09-03 by teachingtechYT