teachingtechYT/teachingtechYT.github.io #114: Feature Request - Bed Level Offsets from edges

rosskinard, 2020-08-28:
Really like the bed-leveling test code, but I use borosilicate glass held on with Swiss clips on my stock Ender 3 and with the standard binder clips on my modified ender 3 that was extended to S4 size (410x410)

It would be nice to be able to enter an offset to basically push the corners inward a bit, since I lose about 15mm at the front and back so the nozzle AND cooling fan ducts clear the clips.

Have an X and Y offset that I could set to something like 20 that would move the squares inward by 20mm in each direction.

teachingtechYT, 2020-08-28:
I already use an offset variable so I don't think it will be hard to implement a user input that is added to this. Look out for this next week.

teachingtechYT, 2020-08-30:
Extra margin from edge added and tested as working on rectangular and circular beds.

closed 2020-08-30 by teachingtechYT