teachingtechYT/teachingtechYT.github.io #116: First Layer Height

ToyMaker62, 2020-08-29:
Can the first layer code generator possibly be for a layer height other than 0.20mm?

This is perhaps more of a newbie question, but if I want to print at 0.28mm or 0.32mm, I don't think I should set my gap using a 0.20mm layer height.  If I did, I would get way too much squish.  Am I suppose to set my slicer to print the first layer at 0.20mm and subsequent layers to 0.32mm?

The same goes the other way, if I want very fine resolution and set my layer height to 0.12mm, I don't believe the filament will even stick to the bed at 0.12mm.

Incidentally, Chuck's boxes use a layer height of 0.28mm.  I've used those to level the bed, get nice adhesion and then printed at 0.16mm and have adhesion problems.  Does he use a first layer height of 0.28mm on all his prints?

Does this make sense or am I lacking the understanding of how layer heights work?

teachingtechYT, 2020-08-29:
I think you are correct. The problem is finding a compromise between simplicity of interface vs power of function.
It's possible this could be implemented but it will become messy. Especially when additional nozzle sizes are added later on.

teachingtechYT, 2021-07-13:
Will be added this week. Only for the first layer test however. Support for multiple nozzles and layer heights is being added universally but there are simply too many permutations to add support for multiple first layer heights within these as well.

closed 2021-07-13 by teachingtechYT