teachingtechYT/teachingtechYT.github.io #120: [FR] Part cooling fan speed tower

teachingtechYT, 2020-08-30:
On behalf of my patron Bob who has even modeled the geometry ready for the test: https://www.prusaprinters.org/prints/38850-5-tiered-overhang-test
The aim is to see the optimum M106 S value for part cooling. Some users upgrade to twin fans with the aim of running them at reduced speed and noise. This test would find a minimum threshold for effective cooling.

sigginet, 2020-09-19:
This is a pretty good idea - I am about to install a hero me 5 with dual blowers.

I'll see if I have time to create a PR for this feature (depends on when my son wakes up)

petemagnusson, 2020-11-15:
This was just what i was looking for 👍

lurkus, 2021-03-27:
This would also be helpful for tuning fan speeds when printing ABS.

Rudis1261, 2021-09-07:
👍  For this. Really struggling with ABS at the moment. This would be a great help to fix that

MartinVoelz, 2022-03-22:
If you program the code, please consider some kind of "break lose torque" setting.
For my yery cheap "toy"   Easythreed K1   I had issues that with

```
M106 0
...
M106 128
```

The fan was not starting to rotate, but with 

```
M106 255
...
M106 128
```

It kept rotating with lower speed.