teachingtechYT/teachingtechYT.github.io #121: Feature Request - Temp Tower and Retraction - Setting for optional larger first layer height

djcosmic22, 2020-08-30:
Would it be possible to add a toggle to retraction and temp tower to make the first layer height be .28? I have noticed I cannot print PETG without this larger first layer height as usually, PETG needs a different amount of squish.

teachingtechYT, 2020-08-30:
This one is very technically hard to implement. The source gcode is pre-sliced and then the site uses Javascript to alter key values on the fly. Even if you substituted in gcode for a thicker first layer, the z values for all of the layers after will be off. I'm afraid the way the site works doesn't leave much flexibility for this type of thing, although I acknowledge in situations like yours it would be useful.

teachingtechYT, 2021-07-13:
I am most of the way through a rewrite of the gcode generators to add support for multiple nozzle widths and layer heights. Adding support for first layer variation for each of these is a bridge too far. I would suggest that the first layer doesn't need to look pretty, just be functional. So hopefully you can get your part to stick by baby stepping up or down because after that is the important stuff.

closed 2021-07-13 by teachingtechYT