teachingtechYT/teachingtechYT.github.io #124: E-Steps Calibration Clarity/Investigation Needed

TheAndr0id, 2020-08-30:
I just completed the E-Steps calibration and after doing some reading on suggested practices, there seems to be some inconsistencies on weather to feed through the hotend or not.

I do see the "Special Note" on the page stating that this should be done without the hotend, but after reviewing multiple guides on this, most are recommending feeding through the hotend.  (and at least with the Ender line you still have to heat the hotend up to run the extruder)

I'm using an Ender 3V2, and when I did the measurements (granted my measurements have inconsistencies as well) I noticed that the same setting had different results if I disconnected the bowden tube vs feeding through the hotend.  I did multiple (6) tests for each.  The following numbers are the averages of those results.

My setting when I did this was 97.28, not the standard 93.00 from Creality.  (Yes, this is round 2)
Cold Extrusion (disconnected bowden tube)     101.44
Hot Extrusion                                                        98.44

3mm of difference.  I have set my new E-Steps using the 98.44 value (M92 E98.82) and remeasured(feeding though the hotend) and it looks spot on.  I do like what I'm seeing so far for new prints (redoing my flow rate now).  The higher number from the cold extrusion makes sense, but is that where we want to measure?  If the slicer says extrude 100mm, is it 100mm of filament, or 100mm of material on the print bed?  (I'm not trying to be sarcastic, I really don't know...)

I'm going to keep an eye on this, but maybe someone could confirm my results/investigate this further.

Is this something that can be clarified from the manufacturers?  As someone who's new to the 3D printing world, this is rather confusing, frustrating and time consuming.

teachingtechYT, 2020-08-30:
I've always done it through the hot end. However, people complained about the documentation so I added the special note.
I think the best person to clarify would be a Marlin dev, as they understand best the relation between E-steps and things that need it to be spot on like linear advance.

teachingtechYT, 2020-09-03:
This one is stuck until we have clarity. Happy to update the instructions if such information becomes available.

closed 2020-09-03 by teachingtechYT