teachingtechYT/teachingtechYT.github.io #125: Slicer Flow Calibration - Prusa Slicer Settings missing

TheAndr0id, 2020-08-30:
In the guide:

Print settings > Advanced > Extrusion width > Default extrusion width: 0.4
and
Print settings > Advanced > Extrusion width > Perimeters: 0.4

Also the External perimeters at least should be changed to 0.4mm.  I changed all of the Extrusion width parameters to 0.4mm just to be sure.

Prusa's Calibration Procedure (https://help.prusa3d.com/en/article/extrusion-multiplier-calibration_2257) recommends using "vase mode" as well.  They do their test using the default 0.45mm extrusion width (which I think is wrong for a 0.4mm nozzle).

Before I made this change I was getting "weird" inconsistent results.  After changing the other Extrusion width parameters in Prusa to 0.4, I'm now getting solid 0.40 measurements.

teachingtechYT, 2020-08-30:
Forgive me but I'm a little confused on the proposed changes. Is this correct:

1. Add instruction for external perimeters to 0.4mm (step 4 for Prusaslicer).
2. Add a new step between 6 and 7 to suggest vase mode.

As for the 0.4 width, I went with this because that's the default for Cura for a 0.2 layer height. I typically use the default 0.48 from Simplify3D. There's a note already explaining the user can substitute whatever value they want here.

TheAndr0id, 2020-08-31:
I think that's kind of what I was thinking.  I'm not sure on how many of the other Extrusion Width options need to be changed as well.  Currently in Prusaslicer 2.2.0 there are the following:
- Default extrusion width
- First layer
- Perimeters
- External perimeters
- Infill
- Solid infill
- Top solid infill
- Support material

I don't think the last three(four?) have any impact on the test cube, but the first four maybe.  When I did my test I changed them all to 0.40 mm.

There is also an "Infill/perimeters overlap" setting which could impact anyone printing 2 wall tests.  Many tests use two walls and I've read on many other sites as well that isn't a good idea.

The Vase mode just seemed to speed things up since there are no seams in the print.   I was just following the suggestion of the Prusaslicer documentation.

As for the 0.4 width I had read that you wanted it to match your nozzle size.  I kind of think this makes sense, but I don't have anything to back that other than other calibration guides out there.

I hope that I'm not coming across as any sort of expert.  I'm just parroting a bunch of stuff I've read and after spending most of the day printing open ended cubes(I have 10 now) I thought I'd pass on what I think I've learned.  

I do appreciate the effort here and was hoping to pass on some extra insight (as long as it *is* insight) to others.   I hope my Issue submissions are not taken as criticism of an excellent resource!

teachingtechYT, 2020-08-31:
Thanks for taking the time to post and document, I appreciate it. The tone reads well to me, don't worry.

My take on the layer width is to view in relation to the ratio to layer height. The nozzle size is important too as a limiting factor. The shape of the extrusion should be mostly rectangular. If the layer width is too narrow compared to the layer height, there won't be enough squish and the extrusion will stay round. That's my take anyway.

I've added external perimeters, that one makes sense.
I've left out first layer since no one will be measuring there.
Add vase mode too as it also makes sense.
Since my instructions are for a single wall cube I won't expand beyond what is already on the note about multi wall cubes.

Thanks again.

closed 2020-08-31 by teachingtechYT