teachingtechYT/teachingtechYT.github.io #126: Minimum line lenght

WiseroOrb, 2020-08-30:
Hi,
when slicing detailed models, the gcode contains a lot of tiny lines, that are printed so fast that the command buffer empties.
this causes the print to stop causing blobs on the surface. However those blobs are detrimental especially for those detailed models.

I wrote a little python script that generates gcode to stress the printer with progressively shorter lines, printing the lenght in the lcd. It does not extrude any filament, and it's quite fast.

When you find the minimum lenght for witch the movement is smooth, you can enter it in your favorite slicer. In Cura is the 'maximum resolution'

For now i only tested it on my ender 3, finding a minimum lenght of 0.8mm

[gen-calibrator.zip](https://github.com/teachingtechYT/teachingtechYT.github.io/files/5147528/gen-calibrator.zip)

teachingtechYT, 2020-08-30:
Are you suggesting this for a new tab?
Sounds like a great idea if so. 
It would need to be converted to run in Javascript from a form in the browser.
To interpret, is the user watching for when the printer starts stuttering/pausing and then taking note of the M117 value on the screen?

WiseroOrb, 2020-08-30:
You interpreted it right. I find it the only way to get the value.

Sadly i know little to nothing of javascript, but i think the conversion is straightforward.

teachingtechYT, 2020-08-31:
I think I should be able to handle the conversion. I'm not great with python but code is code unless there is something really specific to a certain language.
Thanks for the contribution, I'll put this on the to do list.

WiseroOrb, 2020-08-31:
I attach a before and after to show the expected improvements

[image]

teachingtechYT, 2020-09-01:
I ported the code today and am able to recreate the gcode made by your script. It will take a little longer to write the page and record an example of usage.
Thanks for creating this, it is a really innovative feature.

WiseroOrb, 2020-09-01:
Thank you for all your work.

It can be helpful to add a note to test it both printing with Octoprint or from the SD card.
Sometimes printer boards doesn't support fast enough serial connection, causing a bottleneck, therefore longer min-lines.
On the Ender 3 this can be solved flashing a firmware with increased baudrate, in my case from 115.200 to 1.000.000.

teachingtechYT, 2021-07-13:
This is still in my mind. There has been a big rewrite this week that paves the way for future tabs to be added.