teachingtechYT/teachingtechYT.github.io #127: Review Retraction Tuning suggestion - Prahjister's  Retraction Calibration Tool (perhaps implement)

TheAndr0id, 2020-08-31:
After reading the suggestion I took a look at the suggested tool and it looked like a good idea.  I went through the options, generated the GCODE and fired it up.

Then my extruder started making horrific sounds.

The tool generates code which feeds filament far too fast for (at least) my Ender 3V2 to deal with causing the extruder to skip.  I edited the code for the print speed thinking that would help (took it down to 20mm for the first layer) but that worked for about 25% of the base before the extruder started skipping again.  It also made a hell of a mess of the first layer.  I didn't spend any more time digging into the code after that.

This could be a case of his tool is broken, or it could just be not compatible with Enders.  I just know I wasn't happy to be hearing the loud clunk/bang noises coming from my printer...

I've written enough assembly/machine code in my life to be able to understand what is going on in GCODE, but I have zero interest in learning it.  After my experience I wouldn't recommend Prahjister's tool to others and perhaps having this as a suggestion isn't a good idea?  Maybe his code could be forked and fixed?

Then again I could be the only one having issues with his generator...

teachingtechYT, 2020-08-31:
I'm sorry to hear about your troubles. Prahjister posted the link himself here and I added it to the page.
I'm not interested in porting his code but I will add a warning to the link.

teachingtechYT, 2020-08-31:
Done.

closed 2020-08-31 by teachingtechYT

prahjister, 2020-09-09:
Not really sure why the issues with the tool. I have a few theories. I suspect you may have altered the start gcode or restarted a print after a fail. I use a mixture of relative and absolute movements. Which can cause some wacky behavior.

TheAndr0id, 2020-09-09:
I did add some start gcode to prime the nozzle.  Straight up copy n paste from Prusaslicer.    Here's what I put in:

```
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S210 ; set extruder temp
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S210 ; wait for extruder temp
G28 ; home all
G1 Z2 F240
G1 X2 Y10 F3000
G1 Z0.28 F240
G92 E0.0
G1 Y190 E15.0 F1500.0 ; intro line
G1 X2.3 F5000
G1 Y10 E30 F1200.0 ; intro line
G92 E0.0
```

Maybe if I add a G91 to the end?  That would put it back into relative mode, but if your code needs to be in G91 mode it should do that after the start code shouldn't it?  I'll try the tool again without any start gcode.

prahjister, 2020-09-09:
The custom start is really only meant for for autobed leveling or something I couldn't think of. I will clarify the verbiage tonight.

TheAndr0id, 2020-09-14:
The start code above was the issue.  I've run the default (changing only the start temperature and the extrusion multiplier) and this test printed as expected.  I'm now running it again with closer parameters to what I think my Ender 3v2 needs. (Looks like retraction speed isn't an issue at 40mm/s, upping to 20mm/s to see what happens.  5-7mm retraction distance looks like the range)

So as a suggestion to @prahjister, you might want to make sure you set the correct mode after the start code to make sure your test is in the state it thinks it should be.  Setting G90/G91, M82/M83 and G20/G21 at the start of your code would be a good thing.

Otherwise, I would say we've ironed out what's going on here.

prahjister, 2020-09-15:
I will test adding absolute to the beginning and not assume it is already and see how it goes.

As far as extrusion multiplier. I recommend using the defaults in the tool. I used a formula that I found to calculate but after analyzing cura I increased the extrusion multiplier to mimic what cura calls for a 10mm move by default.

prahjister, 2020-09-16:
G90 has been added. Thank you.