teachingtechYT/teachingtechYT.github.io #128: only appears to work with PLA

curtisembedded, 2020-08-31:
Hello,

It is my assumption that these calibration parameters may be unique to each material type. I have successfully used this site to calibrate PLA. However when I try to generate for ABS is appears the Z offset between layers is off? After the first 3-4 layers, the tip is too low and it dislodges the print from the bed.

1. Is it true that the calibrations may be different for different filament types?
2. How do you adjust for the printing thickness of the different materials?

thanks

teachingtechYT, 2020-08-31:
It's true that there are differences in materials and how they print, but I wouldn't say that it limits the functionality of the site.

For instance, when reviewing a new printer without a slicing profile available, I'll often just change temperatures and fan settings from my PLA baseline with good success.

Some things you can try:
Ticking the box for start extra gcode, which will give you an instruction about how to up the flow rate for the print.
Having the part cooling fan on at 50%, which is an option from the drop down.

curtisembedded, 2020-09-01:
Should the instructions state that “for best results, use PLA”?

teachingtechYT, 2020-09-01:
I don't think so as it's not intended to be PLA specific.

closed 2020-09-03 by teachingtechYT