teachingtechYT/teachingtechYT.github.io #130: Bug on calibration test

dennisrosa, 2020-09-01:
Extra restart distance used as unretraction distance.
That has caused a extremely under extrusion problem.

[image]

I sent a PR #129 with the correction of this problem

teachingtechYT, 2020-09-01:
The print is a mess, I agree. However, the code is working as it should be because the start gcode puts the extruder in absolute mode: M82.
Before the retraction sequence, the E distance is reset with G92 E0. Then a retract, eg G1 E-6 F2400
For unretraction, currently the code returns the extruder back to whatever is entered as the extra restart distance by the user.
So if 0 is entered, G1 E0 F2400 will reverse the 6 mm of retraction from my example.
If 0.4 is entered, G1 E0.4 F2400 will unretract 6.4 mm, which is an extra restart of 0.4 mm as desired.
Your proposed code adds the retraction and extra restart, so in this example would give G1 E6.4 F2400. Since the old position was -6, that is 12.4 mm of filament extruded, an extra restart of 6.4 instead of 0.4 mm.
This would only work if the extruder was in relative mode with M83. Did you insert this manually in the custom start gcode field or does your firmware not support M82?
Although the vertical hole where the retraction takes place is prominent in your print, there are also large gaps where the base infill meets the perimeters. Something else not right?

dennisrosa, 2020-09-01:
Tks for you attention.  I understood. 
I'll close the issue.

closed 2020-09-01 by dennisrosa